zephyrproject-rtos/zephyr #28898: lwm2m_client can't start if mcuboot enabled

retfie, 2020-10-04:
**Describe the bug**
lwm2m_client can't start if build with CONFIG_BOOTLOADER_MCUBOOT=y

**To Reproduce**
`cd zephyrproject`
`
west build -b nrf52840dk_nrf52840 -s bootloader/mcuboot/boot/zephyr -d build-mcuboot
`
`
west flash -d build-mcuboot
`
`west build -b nrf52840dk_nrf52840 -s zephyr/samples/net/lwm2m_client -- -DCONFIG_BOOTLOADER_MCUBOOT=y 
-DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem\"
`
`
west flash
`

`
*** Booting Zephyr OS build zephyr-v2.4.0-64-g20df3ffc3dcd  ***
`
`
[00:00:00.269,836] <inf> mcuboot: Starting bootloader
`
`
[00:00:00.276,306] <inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
`
`
[00:00:00.286,590] <inf> mcuboot: Boot source: none
`
`
[00:00:00.292,236] <inf> mcuboot: Swap type: none
`
`
[00:00:00.377,624] <inf> mcuboot: Bootloader chainload address offset: 0xc000
`
`
[00:00:00.385,528] <inf> mcuboot: Jumping to the first image slot
`

**Expected behavior**
After control is passed to first image slot, APP main should be started:
`
*** Booting Zephyr OS build zephyr-v2.4.0-64-g20df3ffc3dcd  ***
`
`
00:00:00.008,453] <inf> net_lwm2m_client_app: Run LWM2M client
`

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gcc-arm-none-eabi-7-2018-q2-update
 - Zephyr version: 2.4.99
 - Commit SHA: 20df3ffc3dcd1de5a9ad45d713fc656ea88cad99

**Additional context**
If set CONFIG_LWM2M=n and comment all the code into lwm2m_client.c and left only 
`
LOG_INF(APP_BANNER);
`
into main(), then APP banner is shown.
Increasing CONFIG_MAIN_STACK_SIZE to 2048 or 4096, doesn't help.

retfie, 2020-10-06:
One working solution for now is #28931 by setting CONFIG_HW_STACK_PROTECTION=n in mcuboot according to this PR into mainline mcuboot: JuulLabs-OSS/mcuboot#828. 
With this change I was able to get pretty big example running after control was passed from mcuboot.

nvlsianpu, 2020-10-08:
will be fixed by #28967

PR #28967 (2020-10-09): manifest: synch MCUBoot to upstream of 7 10 2020
Introduces what is in https://github.com/zephyrproject-rtos/mcuboot/pull/36

MCUBoot was synchronized up to:
https://github.com/JuulLabs-OSS/mcuboot/commit/296949e

Improvements:
- hardening against hardware level fault injection and timing attacks
- Abstract crypto primitives to simplify porting.
- boot: Add ram-load upgrade mode
- renamed single-image mode to single-slot mode

- kconfig: provide logic for setting key file, simplify prj.conf

- imgtool: Print image digest during verify
- imgtool: Add possibility to set confirm flag for hex file [... cut]
closed 2020-10-09 by carlescufi