zephyrproject-rtos/zephyr #28899: memory-mapped thread stacks to catch overflows

andrewboie, 2020-10-04:
**Is your enhancement proposal related to a problem? Please describe.**
On MMU devices, we do not memory map stacks, other than their permanent mapping in RAM, which requires a read-only *physical* page to catch overflows to be reserved. This wastes 4K of physical memory per thread stack (see the `guard_page` member of `struct z_x86_thread_stack_header`.

We don't clean these up at thread exit either, which could cause problems if stack memory is re-used for some other purpose, such as being released back to a heap.

**Describe the solution you'd like**
We'd like to add support for memory mapping thread stacks. By memory mapping the thread-accessible buffer in the virtual address space, we can leave a non-present virtual page immediately below it, to catch stack overflows.

The steps to implement this are roughly as follows:
1. We currently can only map memory, but not un-map it. Implement a function to un-map memory.
2. For any given stack buffer we already have a permanent RAM memory mapping for it, but don't use this. On thread creation, memory map the stack buffer (within the larger stack object) in the virtual address space, with a non-present page immediately below it.
3. Set the thread's initial stack pointer within this mapping (the value returned by `setup_thread_stack()`)
4. Do whatever's needed to ensure that the User bit is only set for the target thread or threads within its memory domain.
5. On thread exit, un-map its stack memory, the place to hook this up is `z_thread_single_abort()`
6. Adjust the fatal page fault handling code to check if the faulting address was within the virtual guard page for the thread, and report a stack overflow if so. Otherwise it's just a normal page fault.
7. Remove the physical guard page from the stack object specification, we no longer need it.

All of this is only needed if CONFIG_HW_STACK_PROTECTION is enabled, although this could later be extended to support dynamic stacks that grow.

**Describe alternatives you've considered**
None so far.

dcpleung, 2024-02-12:
This is still useful. Will look into in when time permits.

cfriedt, 2024-04-01:
I love seeing these long-lived tickets getting some attention 🪁

dcpleung, 2024-04-26:
The core kernel support for this is in, and x86 has code to utilize this. Next step is for each architecture to support this if desired.

closed 2024-04-26 by dcpleung