zephyrproject-rtos/zephyr #28916: net_if_down doesn't clear address

endian-benjamin, 2020-10-05:
**Describe the bug**
When trying to implement PPP stop/start with GSM, I noticed that that on the second time we bring up the network interface, the IPCP negotiations go through fine, but the iface address wasn't updated - there is a stale ipv4 unicast address associated with the iface. 

After much GDB fussing, I found that `net_if_ipv4_addr_add` is called, but it can't add the new IP because the old one is still there, marked as `in_use`. 

I've fixed it already as part of #28475 but since that is a WIP I am starting a new PR.

endian-benjamin, 2020-10-05:
See #28917

jukkar, 2020-10-06:
If user/app sets an IP address manually, we should not clear it automatically by `net_if_down()`. The one who set the address is responsible clearing it.

endian-benjamin, 2020-10-06:
Merged

closed 2020-10-06 by endian-benjamin