zephyrproject-rtos/zephyr #28921: MCUboot / smp_svr sample broken in 2.4.0

CodingGhost, 2020-10-05:
**Describe the bug**
since 2.4.0 the smp_svr sample is broken.
I have exactly followed the instructions, but it does not work, its also confirmed by sombody else.
Mcuboot runs fine, but as soon as it jumps to the smp_svr sample its just dead and nothing happens anymore. its also not advertising. This only happens when using the overlay-bt, overlay-bt-tiny works fine. It seems like it has something todo with the image size.
overlay-bt works fine with 2.3.0.

**To Reproduce**
Steps to reproduce the behavior:
just use west init and then west update to get a clean copy, then:

```
west build -b nrf52840dk_nrf52840 bootloader/mcuboot/boot/zephyr
west flash
rm -rf build
west build -b nrf52840dk_nrf52840 zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DOVERLAY_CONFIG=overlay-bt.conf
west sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem
west flash --hex-file build/zephyr/zephyr.signed.hex
```

**Expected behavior**
MCUboot loads the sample and runs

**Impact**
have to go back to 2.3.0

**Logs and console output**
```
*** Booting Zephyr OS build v2.4.0-rc3  ***
[00:00:00.376,495] [1B][0m<inf> mcuboot: Starting bootloader[1B][0m
[00:00:00.383,056] [1B][0m<inf> mcuboot: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3[1B][0m
[00:00:00.393,493] [1B][0m<inf> mcuboot: Boot source: none[1B][0m
[00:00:00.399,261] [1B][0m<inf> mcuboot: Swap type: none[1B][0m
[00:00:00.678,283] [1B][0m<inf> mcuboot: Bootloader chainload address offset: 0xc000[1B][0m
[00:00:00.686,309] [1B][0m<inf> mcuboot: Jumping to the first image `slot[1B][0m`
```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain Zephyr SDK

**Additional context**
It seems like the problem has something todo with the image size, as I tried one of my own apps, and it also doesnt work (its way bigger than the sample)

mniestroj, 2020-10-05:
Bisect shows:
```
ded737cd5829414b2126a2e9a09dceacd20d1938 is the first bad commit
commit ded737cd5829414b2126a2e9a09dceacd20d1938
Author: Carles Cufi <carles.cufi@nordicsemi.no>
Date:   Thu Sep 17 15:44:16 2020 +0200
    boards: nordic: Turn hardware stack protection on by default
    
    In all Nordic boards that contain ICs that support hardware stack
    protection, turn it on by default so that we detect
    stack overflows easier.
    
    Signed-off-by: Carles Cufi <carles.cufi@nordicsemi.no>
 boards/arm/nrf21540dk_nrf52840/nrf21540dk_nrf52840_defconfig         | 3 +++
 boards/arm/nrf52833dk_nrf52820/nrf52833dk_nrf52820_defconfig         | 3 +++
 boards/arm/nrf52833dk_nrf52833/nrf52833dk_nrf52833_defconfig         | 3 +++
 boards/arm/nrf52840dk_nrf52811/nrf52840dk_nrf52811_defconfig         | 3 +++
 boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig         | 3 +++
 boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840_defconfig | 3 +++
 boards/arm/nrf52dk_nrf52805/nrf52dk_nrf52805_defconfig               | 3 +++
 boards/arm/nrf52dk_nrf52810/nrf52dk_nrf52810_defconfig               | 3 +++
 boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832_defconfig               | 3 +++
 boards/arm/nrf5340dk_nrf5340/nrf5340dk_nrf5340_cpuapp_defconfig      | 3 +++
 boards/arm/nrf5340dk_nrf5340/nrf5340dk_nrf5340_cpuappns_defconfig    | 3 +++
 boards/arm/nrf5340dk_nrf5340/nrf5340dk_nrf5340_cpunet_defconfig      | 3 +++
 boards/arm/nrf5340dk_nrf5340/nrf5340pdk_nrf5340_cpuapp_defconfig     | 3 +++
 boards/arm/nrf5340dk_nrf5340/nrf5340pdk_nrf5340_cpuappns_defconfig   | 3 +++
 boards/arm/nrf5340dk_nrf5340/nrf5340pdk_nrf5340_cpunet_defconfig     | 3 +++
 boards/arm/nrf9160dk_nrf52840/nrf9160dk_nrf52840_defconfig           | 3 +++
 boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_defconfig             | 3 +++
 boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160ns_defconfig           | 3 +++
 boards/arm/thingy52_nrf52832/thingy52_nrf52832_defconfig             | 3 +++
 19 files changed, 57 insertions(+)
```

It turns out that stack protection is enabled by default in mcuboot after this commit. Then, after we jump from bootloader to application code, we end up being "catched" by hw stack protection mechanism.

The solution for now seems to be setting `CONFIG_HW_STACK_PROTECTION=n` in mcuboot. Here is PR to mainline mcuboot: https://github.com/JuulLabs-OSS/mcuboot/pull/828.

JordanYates, 2020-10-06:
The actual error condition (for my applications at least).

`CONFIG_HW_STACK_PROTECTION`  changes the default on `ARM_STACK_PROTECTION`, which selects `MPU_STACK_GUARD`.
`MPU_STACK_GUARD` sets a region towards the start of RAM to be Read-Only in the bootloader when the thread is loaded.
This region is still configured when the application is jumped into.

`z_bss_zero` in `z_cstart` attempts to zero the region, resulting in a 'Data Access Violation'.
I can confirm that forcing `CONFIG_HW_STACK_PROTECTION=n` fixes the issue.

PR #28967 (2020-10-09): manifest: synch MCUBoot to upstream of 7 10 2020
Introduces what is in https://github.com/zephyrproject-rtos/mcuboot/pull/36

MCUBoot was synchronized up to:
https://github.com/JuulLabs-OSS/mcuboot/commit/296949e

Improvements:
- hardening against hardware level fault injection and timing attacks
- Abstract crypto primitives to simplify porting.
- boot: Add ram-load upgrade mode
- renamed single-image mode to single-slot mode

- kconfig: provide logic for setting key file, simplify prj.conf

- imgtool: Print image digest during verify
- imgtool: Add possibility to set confirm flag for hex file [... cut]
closed 2020-10-09 by carlescufi