zephyrproject-rtos/zephyr #28925: west failed due to empty value in self.path

sdalu, 2020-10-05:
Tested with west 0.8.0 on FreeBSD system

If self.path is changed to use manifest-repo, everything is working fine

Error message
~~~
-- west build: generating a build system
-- Application: ....
-- Zephyr version: 2.4.0 (/home/sdalu/ZephyrProjects/zephyr)
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
CMake Error at /home/sdalu/ZephyrProjects/zephyr/cmake/zephyr_module.cmake:48 (message):
  Traceback (most recent call last):

    File "/home/sdalu/ZephyrProjects/zephyr/scripts/zephyr_module.py", line 326, in <module>
      main()
    File "/home/sdalu/ZephyrProjects/zephyr/scripts/zephyr_module.py", line 266, in main
      meta = process_module(project)
    File "/home/sdalu/ZephyrProjects/zephyr/scripts/zephyr_module.py", line 99, in process_module
      module_path = PurePath(module)
    File "/usr/local/lib/python3.7/pathlib.py", line 642, in __new__
      return cls._from_parts(args)
    File "/usr/local/lib/python3.7/pathlib.py", line 674, in _from_parts
      drv, root, parts = self._parse_args(args)
    File "/usr/local/lib/python3.7/pathlib.py", line 658, in _parse_args
      a = os.fspath(a)

  TypeError: expected str, bytes or os.PathLike object, not NoneType

Call Stack (most recent call first):
  /home/sdalu/ZephyrProjects/zephyr/cmake/app/boilerplate.cmake:150 (include)
  CMakeLists.txt:6 (include)
~~~

Manifest used in T3 configuration. manifest is put in it's own `manifest-repo` directory
~~~yaml
manifest:
  remotes:
    - name: zephyrproject-rtos
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      remote: zephyrproject-rtos
      revision: v2.4.0
      import: true    

  self:
    path:
~~~

`.west/config` file
~~~yaml
[manifest]
path = manifest-repo
[zephyr]
base = zephyr
~~~

mbolivar-nordic, 2020-10-05:
```yaml
  self:
    path:
```

Can you explain why this is here? The semantics of `self: path: foo` are to place the manifest repository at path `foo` during `west init`, so leaving this as an empty string is not going to lead to good results.

Do you still see this if you delete this entire `self:` map?

mbolivar-nordic, 2020-10-05:
@tejlmand either way I think we need to avoid a stack trace here. Need to see if it's a zephyr_module.py issue or if west.manifest should reject this manifest completely

sdalu, 2020-10-05:
Building fine without the `self:` section.

For the:
~~~
  self:
    path:
~~~
I don't remember exactly where I got it but I think it was the recommended way in previous (2.0 ?) documentation for T3 forest topology

mbolivar-nordic, 2020-10-05:
> I don't remember exactly where I got it but I think it was the recommended way in previous (2.0 ?) documentation for T3 forest topology, and it was working fine that way before 2.4

This manifest is not well-formed; if you can point to some documentation that recommends doing it that way, I will fix it. I don't see any in master.

I think this is therefore not a bug, but a malformed manifest. I've sent a PR to west which rejects this kind of manifest so it doesn't cause confusion in the future: https://github.com/zephyrproject-rtos/west/pull/446

mbolivar-nordic, 2020-10-05:
Since https://github.com/zephyrproject-rtos/west/pull/446 is merged, I'm going to close this. I think I've answered the questions about best practices around `self: path:` in the PR comments there as well -- there's no "one size fits all" answer; it's really up to the user. Please LMK if you think any docs are still missing and I will send an update.

closed 2020-10-05 by mbolivar-nordic