zephyrproject-rtos/zephyr #28933: mcuboot: Brick when using BOOT_SWAP_USING_MOVE and reset happens during images swap

mniestroj, 2020-10-05:
**Describe the bug**
Interrupted swap of images when **swap_move** (`CONFIG_BOOT_SWAP_USING_MOVE=y`) strategy (default for nRF chips) is selected, can end up in "bricked" state. Possible workaround is to use **swap_scratch** (`CONFIG_BOOT_SWAP_USING_SCRATCH=y`) strategy, which doesn't seem to have this specific issue.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -d build_boot -b nrf52840dk_nrf52840 ../bootloader/mcuboot/boot/zephyr/`
2. `west flash -d build_boot`
3. `west build -b nrf52840dk_nrf52840 samples/subsys/mgmt/mcumgr/smp_svr/ -- -DOVERLAY_CONFIG=overlay-bt.conf`
4. `west sign -t imgtool --no-bin -H app.hex -- --key ../bootloader/mcuboot/root-rsa-2048.pem --version 1.0.0 --pad --confirm`
5. `west flash --hex-file app.hex`
6. `west sign -t imgtool --no-hex -B fake.bin -- --key ../bootloader/mcuboot/root-rsa-2048.pem --version 1.2.3`
7. `mcumgr --conntype ble --connstring peer_name=Zephyr image upload fake.bin`
8. `mcumgr --conntype ble --connstring peer_name=Zephyr image test 597f80daf86f4d5bbe0a23af28167ae1de07cbd441b1c2db862e8dff17cf2037`
```
Images:
 image=0 slot=0
    version: 1.0.0
    bootable: true
    flags: active confirmed
    hash: 096918ac22fbc49f7613ccabbd06853804c7422a7884239c212c7f70dd58976e
 image=0 slot=1
    version: 1.2.3
    bootable: true
    flags: pending
    hash: 597f80daf86f4d5bbe0a23af28167ae1de07cbd441b1c2db862e8dff17cf2037
Split status: N/A (0)
```
9. Press RESET button, wait ~2s, press RESET.
Log shows mcuboot messages after first (and second) reset:
```
$ cat /dev/ttyACM1 | ts '[%H:%M:%.S]'
[20:14:48.323953] *** Booting Zephyr OS build zephyr-v2.4.0-226-gb89548198d99  ***
[20:14:48.329925] [00:00:00.257,659] <inf> mcuboot: Starting bootloader
[20:14:48.339929] [00:00:00.264,129] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
[20:14:48.346898] [00:00:00.274,414] <inf> mcuboot: Boot source: primary slot
[20:14:48.355929] [00:00:00.283,966] <inf> mcuboot: Swap type: test
[20:14:50.282960] *** Booting Zephyr OS build zephyr-v2.4.0-226-gb89548198d99  ***
[20:14:50.288973] [00:00:00.257,598] <inf> mcuboot: Starting bootloader
[20:14:50.298939] [00:00:00.264,068] <inf> mcuboot: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
[20:14:50.305945] [00:00:00.274,322] <inf> mcuboot: Boot source: primary slot
```
10. Board never recovers.

**Expected behavior**
Swap of images should continue.

**Impact**
Remote firmware upgrade is not safe, i.e. can result in bricked devices in the field.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.11.4
 - Commit SHA: dac6937cdacd4473f3e6d16964436cbf8c25b295 (2.4.99) + https://github.com/JuulLabs-OSS/mcuboot/pull/828 bugfix

**Additional context**
It seems that problems start here (https://github.com/JuulLabs-OSS/mcuboot/blob/aac99a14c9cdc8684fd4fbced130d3065a75ffea/boot/bootutil/src/loader.c#L1084):
```
        /*
         * If a swap was under way, the swap_size should already be present
         * in the trailer...
         */
        rc = boot_read_swap_size(image_index, &bs->swap_size);
        assert(rc == 0);
```
mcuboot reads `0xFFFFFFFF` after swap was interrupted, because this is what was contained originally in `app.hex` (signed and confirmed) image trailer and that was never updated by mcuboot. This value is passed as `copy_size` to `swap_run()` and results in infinite loop there (https://github.com/JuulLabs-OSS/mcuboot/blob/aac99a14c9cdc8684fd4fbced130d3065a75ffea/boot/bootutil/src/swap_move.c#L448):
```
    while (1) {
        sz += sector_sz;
        /* Skip to next sector because all sectors will be moved up. */
        g_last_idx++;
        if (sz >= copy_size) {
            break;
        }
    }
```

utzig, 2020-10-06:
I was able to reproduce what I suppose is a similar issue (also bricking) on the `frdm_k64f` but only seems to happen when using `--pad --confirm` in step 4; padding+confirming the image in the primary slot is completely unnecessary, would you mind giving a try without passing those flags? In the interim I'll check if I can reproduce the issue in the simulator, so hopefully it will be easy to fix.

mniestroj, 2020-10-07:
>padding+confirming the image in the primary slot is completely unnecessary, would you mind giving a try without passing those flags?

I confirm that everythink is fine without using `--confirmed --pad` flags, when flashing to fully erased slot.

The reason why I am using padded images is that I don't know whether there was some data on the (primary) slot to be flashed. Padded image makes sure it will replace all contents of image slot, because flash operation is prededed by erase of region where image lies.

PR #28967 (2020-10-09): manifest: synch MCUBoot to upstream of 7 10 2020
Introduces what is in https://github.com/zephyrproject-rtos/mcuboot/pull/36

MCUBoot was synchronized up to:
https://github.com/JuulLabs-OSS/mcuboot/commit/296949e

Improvements:
- hardening against hardware level fault injection and timing attacks
- Abstract crypto primitives to simplify porting.
- boot: Add ram-load upgrade mode
- renamed single-image mode to single-slot mode

- kconfig: provide logic for setting key file, simplify prj.conf

- imgtool: Print image digest during verify
- imgtool: Add possibility to set confirm flag for hex file [... cut]
closed 2020-10-09 by carlescufi