zephyrproject-rtos/zephyr #28935: support code coverage in unit tests

pabigot, 2020-10-05:
Coverage of standard tests can be done with a command like:

    sanitycheck --platform native_posix_64 -C -T tests/lib/onoff/

This doesn't seem to work for unit tests:

```
tirzah[93]$ sanitycheck --platform native_posix_64 -C -T tests/unit/timeutil/
Renaming output directory to /mnt/devel/zp/zephyr/sanity-out.8
INFO    - JOBS: 32
INFO    - Building initial testcase list...
INFO    - 0 test configurations selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...

INFO    - 0 of 0 tests passed (0.00%), 0 failed, 0 skipped with 0 warnings in 0.31 seconds
INFO    - In total 0 test cases were executed on 0 out of total 289 platforms (0.00%)
INFO    - 0 tests executed on platforms, 0 tests were only built.
INFO    - Generating coverage files...
geninfo: WARNING: no .gcda files found in /mnt/devel/zp/zephyr/sanity-out - skipping!
lcov: ERROR: no valid records found in tracefile /mnt/devel/zp/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /mnt/devel/zp/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /mnt/devel/zp/zephyr/sanity-out/coverage.info
lcov: ERROR: no valid records found in tracefile /mnt/devel/zp/zephyr/sanity-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /mnt/devel/zp/zephyr/sanity-out/coverage.info
tirzah[94]$
```

It'd be nice to have coverage for these tests.

pabigot, 2020-10-05:
Not sure who should take this.

nashif, 2020-10-06:
```sanitycheck -p unit_testing -T tests/unit/crc --coverage```

nashif, 2020-10-06:
unit tests are not run with native_posix, they are built as host binaries, you need to use special board 'unit_testing' for this.

pabigot, 2020-10-06:
OK, confirmed that this works.  Seems like this should be documented somewhere, probably in `coverage.rst`.  Do you want to do that @nashif or should I do it?

nashif, 2020-10-06:
> Do you want to do that @nashif or should I do it?

be my guest.

PR #28968 (2020-10-06): doc: coverage: document unit test coverage
Unit tests require a special board and use the host toolchain.

Closes #28935.
closed 2020-10-06 by MaureenHelm