zephyrproject-rtos/zephyr #28941: Civetweb: create separate directory

Nukersson, 2020-10-05:
**Is your feature request related to a problem? Please describe.**
CivetWeb using samples have potential to spread over multiple directories (especially after merging of #28323).

**Describe the solution you'd like**
Put all CivetWeb using samples into one `civetweb` root directory. Thus allowing http/websocket and many more CivetWeb based samples existing in one root directory (e.g. `samples/net/civetweb/http_server`, `samples/net/civetweb/websocket_server`).

**Describe alternatives you've considered**
none. (Maybe your suggestion?)

**Additional context**
Should be done incrementally after merging of #28323

Nukersson, 2020-10-05:
@jukkar It was mostly your idea, to put CivetWeb samples into same root directory. Please provide short look at this issue.

jukkar, 2020-10-06:
Agreed with separate directory. The `samples/net/civetweb/` is good place to put the sample apps.

Nukersson, 2020-11-21:
Reopened due to  #30175

closed 2020-11-24 by jukkar