zephyrproject-rtos/zephyr #28942: ARC: nsim_hs_smp: huge zephyr.hex file generated on build

evgeniy-paltsev, 2020-10-05:
**Describe the bug**
Build for ARC nsim_hs_smp board generates huge zephyr.hex file (5.7G for sample.basic.helloworld)
Most likely it is caused by big empty hole in memory layout.

We should to check if we can disable zephyr.hex generation in case of nsim_hs_smp or adjust memory layout.

The file is generated even if it is disabled in .config:
```
$ cat build/zephyr/.config | grep "CONFIG_BUILD_OUTPUT_HEX"
# CONFIG_BUILD_OUTPUT_HEX is not set
```

**To Reproduce**
Steps to reproduce the behavior:
```
west build -b nsim_hs_smp samples/hello_world
du -sh build/zephyr/zephyr.hex
```

**Impact**
It bloats your FS (especially in case of sanitycheck)
It dramatically increases compilation time

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.11.4, ARC GNU
 - Commit SHA or Version used: 3ab5ac81ca

IRISZZW, 2020-10-15:
I tried in my VM, and our server.
I can't reproduce it, no zephyr.hex generated.
```
$ git checkout 3ab5ac8 -b huge-hex
$ west build -b nsim_hs_smp samples/hello_world
$ ls build/zephyr/zephyr*
build/zephyr/zephyr.dts  build/zephyr/zephyr.elf  build/zephyr/zephyr.lst  build/zephyr/zephyr.map  build/zephyr/zephyr_prebuilt.elf  build/zephyr/zephyr_prebuilt.map  build/zephyr/zephyr.stat
```

evgeniy-paltsev, 2020-12-15:
The issue is related to openocd presence in custom config, I'll look to it by myself.

nashif, 2021-01-30:
@evgeniy-paltsev update please or close... Cant reproduce here, if this is due to local env then we should close it.

nashif, 2021-02-02:
Can't reproduce.

closed 2021-02-02 by nashif