zephyrproject-rtos/zephyr #28948: hci_usb: ACL transfer not restarted after USB Suspend - Resume

mkarhumaa, 2020-10-06:
**Describe the bug**
hci_usb sample does not tolerate USB Suspend - Resume cycle. USB transfer for ACL data is not restarted after host has done Suspend - Resume to the device.

From USB debug logs we see that when host does Suspend, Zephyr cancels the pending transfer:
```
[00:00:16.481,414] <dbg> usb_transfer.usb_cancel_transfers: Cancel transfer for ep: 0x01
[00:00:16.481,414] <dbg> usb_bluetooth.bluetooth_status_cb: USB device suspended
[00:00:16.481,445] <dbg> usb_transfer.usb_transfer_work: Transfer cancelled or completed, ep 0x01
[00:00:16.481,475] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x01, status -72, size 0
```

Then on Resume, transfer is not started for ep 0x01:
```
[00:00:34.820,861] <dbg> usb_bluetooth.bluetooth_status_cb: USB device resumed
[00:00:34.934,204] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:34.934,234] <dbg> usb_device.custom_handler: bRequest 0x00, wIndex 0x0000
[00:00:34.934,265] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_STATUS
[00:00:34.934,356] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[00:00:34.934,692] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:34.934,875] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:00:34.934,936] <dbg> usb_device.class_handler: bRequest 0x00, wIndex 0x0000
[00:00:34.934,936] <dbg> usb_bluetooth.bluetooth_class_handler: len 3
[00:00:34.934,967] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
[00:00:34.935,241] <dbg> usb_transfer.usb_transfer: Transfer start, ep 0x81, data 0x2000f5c1, dlen 6
[00:00:34.936,218] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x81, status 0, size 6
[00:00:34.936,492] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:00:34.936,706] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
...
```

Then when connecting to another device through bluetoothctl, ACL data is dropped because there was no transfer.

`[00:01:48.974,243] <err> usb_transfer: RX data lost, no transfer`

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/bluetooth/hci_usb
2. west build -b west build -b nrf52840dk_nrf52840 && west flash
3. Start btmon in one terminal, bluetoothctl in another. Run both as root.
4. on bluetoothctl do:
 * power off (then wait more than 3 seconds)
 * power on
5. then scan and try to connect to any device
6. check btmon capture and see that after ACL Data TX `ATT: Exchange MTU Request` there is no `Number of Completed Packets` event and another device does not respond to the 

**Expected behavior**
I would expect Zephyr hci_usb firmware to tolerate USB suspend - resume. USB suspend - resume is done when doing power off - power on on Bluez.

**Impact**
Showstopper

**Logs and console output**
Full output of USB debug log:
```
*** Booting Zephyr OS build v2.4.0-rc2-361-ged7263bf6b91  **** Booting Zephyr OS build v2.4.0-rc2-361-ged7263bf6b91  ***
Bluetooth over USB sample
[00:00:00.251,464] <dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_start 0x20000654
[00:00:00.251,464] <dbg> usb_descriptor.usb_get_device_descriptor: __usb_descriptor_end 0x200006d3
[00:00:00.251,464] <dbg> usb_descriptor.usb_fix_descriptor: Configuration descriptor 0x20000666
[00:00:00.251,464] <dbg> usb_descriptor.usb_fix_descriptor: Interface descriptor 0x2000066f
[00:00:00.251,464] <dbg> usb_descriptor.usb_fix_descriptor: Endpoint descriptor 0x20000678
[00:00:00.251,495] <dbg> usb_descriptor.usb_validate_ep_cfg_data: Fixing EP address 81 -> 81
[00:00:00.251,495] <dbg> usb_descriptor.usb_validate_ep_cfg_data: endpoint 0x81
[00:00:00.251,495] <dbg> usb_descriptor.usb_fix_descriptor: Endpoint descriptor 0x2000067f
[00:00:00.251,495] <dbg> usb_descriptor.usb_validate_ep_cfg_data: Fixing EP address 2 -> 1
[00:00:00.251,495] <dbg> usb_descriptor.usb_validate_ep_cfg_data: endpoint 0x1
[00:00:00.251,525] <dbg> usb_descriptor.usb_fix_descriptor: Endpoint descriptor 0x20000686
[00:00:00.251,525] <dbg> usb_descriptor.usb_validate_ep_cfg_data: Fixing EP address 82 -> 82
[00:00:00.251,525] <dbg> usb_descriptor.usb_validate_ep_cfg_data: endpoint 0x82
[00:00:00.251,525] <dbg> usb_descriptor.usb_fix_descriptor: Now the wTotalLength is 39
[... 621 lines cut ...]
[00:01:49.973,449] <dbg> usb_bluetooth.bluetooth_class_handler: len 17
[00:01:49.973,480] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
[00:01:49.973,571] <dbg> usb_transfer.usb_transfer: Transfer start, ep 0x81, data 0x2000f5c1, dlen 6
[00:01:49.974,670] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x81, status 0, size 6
[00:01:50.320,404] <dbg> usb_transfer.usb_transfer: Transfer start, ep 0x81, data 0x2000f5c1, dlen 12
[00:01:50.320,678] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x81, status 0, size 12
[00:02:20.982,696] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
[00:02:20.982,879] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x01
[00:02:20.982,910] <dbg> usb_device.class_handler: bRequest 0x00, wIndex 0x0000
[00:02:20.982,910] <dbg> usb_bluetooth.bluetooth_class_handler: len 6
[00:02:20.982,940] <dbg> usb_device.usb_handle_control_transfer: >> usb_data_to_host(2)
[00:02:20.983,032] <dbg> usb_transfer.usb_transfer: Transfer start, ep 0x81, data 0x2000f5c1, dlen 6
[00:02:20.983,459] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x81, status 0, size 6
[00:02:21.008,972] <dbg> usb_transfer.usb_transfer: Transfer start, ep 0x81, data 0x2000f5c1, dlen 6
[00:02:21.009,460] <dbg> usb_transfer.usb_transfer_work: Transfer done, ep 0x81, status 0, size 6                                                                    
```

**Environment (please complete the following information):**
 - OS: Linux, Ubuntu 20.04, kernel 5.4.0-48-generic
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: Zephyr master ed7263bf6b91d4542573d2f44ed742a2acf542df. Also 2.2.1 is affected.

**Additional context**

See https://github.com/zephyrproject-rtos/zephyr/pull/28569 for proposed fix and discussion around it.

emob-nordic, 2020-10-06:
@mkarhumaa Please see the linked PR and test if it works for You, please.

mkarhumaa, 2020-10-06:
@emob-nordic Thank you for quick fix. Your fix seems to be working fine. :+1:

PR #28960 (2020-10-20): usb: hci: Handle Suspend->Resume events properly.
USB bus might be suspended to save power. After the device
is Resumed from Suspended state there is a need to restart
OUT transfers for Endpoints used by HCI class. The transfers
shall be restarted only if the device was Resumed after Suspend
from Configured state. This patch applies the fix.

Fixes #28948

Reference: #26725
closed 2020-10-20 by MaureenHelm