zephyrproject-rtos/zephyr #28955: undesired kernel debug log

kurddt, 2020-10-06:
**Describe the bug**
If the default log level is set to debug, there is no way to remove some of the kernel debug log. regardless of the value of `CONFIG_KERNEL_LOG_LEVEL` some kernel debug log are present

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Hello word sample application folder
2. Edit `prj.conf` and add:
```
CONFIG_LOG_DEFAULT_LEVEL=4
CONFIG_KERNEL_LOG_LEVEL_INF=y
```
3. Build & run the sample
4. Some kernel debug log are present (see log output)

**Expected behavior**
In this configuration only kernel information message should be printed as well as user log:
```
*** Booting Zephyr OS build zephyr-v2.4.0  ***
Hello World! nrf52dk_nrf52832
```

**Impact**
This bug is annoying because user log is cluttered with unwanted kernel debug message

**Logs and console output**
```
[00:00:00.412,078] <dbg> os.z_impl_k_mutex_lock: 0x20000140 took mutex 0x20000064, count: 1, orig prio: 0
[00:00:00.422,515] <dbg> os.z_impl_k_mutex_unlock: mutex 0x20000064 lock_count: 1
[00:00:00.430,786] <dbg> os.z_impl_k_mutex_unlock: new owner of mutex 0x20000064: 0x00000000 (prio: -1000)
[00:00:00.441,284] <dbg> os.k_sched_unlock: scheduler unlocked (0x20000140:0)
[00:00:00.449,218] <dbg> os.z_impl_k_mutex_lock: 0x20000140 took mutex 0x20000064, count: 1, orig prio: 0
[00:00:00.459,655] <dbg> os.z_impl_k_mutex_unlock: mutex 0x20000064 lock_count: 1
[00:00:00.467,926] <dbg> os.z_impl_k_mutex_unlock: new owner of mutex 0x20000064: 0x00000000 (prio: -1000)
[00:00:00.478,424] <dbg> os.k_sched_unlock: scheduler unlocked (0x20000140:0)

*** Booting Zephyr OS build zephyr-v2.4.0  ***
[00:00:00.490,661] <dbg> os.k_sched_unlock: scheduler unlocked (0x20000140:0)
Hello World! nrf52dk_nrf52832
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Arm GNU embedded
 - Zephyr 2.4.0 (not present in Zephyr 2.3.0)
 - RTT log backend

CodingGhost, 2020-11-26:
I am experiencing this too, its really annoying.

nashif, 2020-11-26:
@nordic-krch  Can you please take a look at this, can't find what is going on here.

CodingGhost, 2020-11-26:
I think I know what has happened: the default log level kconfig options seems to affect the kernel log level since 2.4.0, maybe this isnt a bug but just a feature thats confusing for people updating to 2.4.0.
setting the default log level to 1 gets rid of the messages

PR #30271 (2020-11-27): kernel: Cleanup logger setup in kernel files
Most of kernel files where declaring os module without providing
log level. Because of that default log level was used instead of
CONFIG_KERNEL_LOG_LEVEL.

Fixes #28955.
closed 2020-11-27 by nashif