zephyrproject-rtos/zephyr #28956: API-less devices aren't findable

endian-benjamin, 2020-10-06:
`device_get_binding(NAME)` can't find devices without API structs defined because of 

```
/** @brief Determine whether a device has been successfully initialized.
 *
 * @param dev pointer to the device in question.
 *
 * @return true if and only if the device is available for use.
 */
static inline bool z_device_ready(const struct device *dev)
{
	return dev->driver_api != NULL;
}
```

This seems like a pretty cheap hack, since all devices do not have a driver API but should still be findable (for instance because they have a power management API). Successfully initialized is NOT equivalent to `driver_api != NULL`.

I currently have no alternative solution to checking if the device was inited properly, but the current solution is not good.

endian-benjamin, 2020-10-06:
Ah, I see this has been modified in latest upstream...

pabigot, 2020-10-06:
Also #28873 is relevant, but I believe this specific concern isn't an issue now.  Re-open if necessary.

closed 2020-10-06 by pabigot