zephyrproject-rtos/zephyr #28959: Bluetooth: controller: Use inline wrapper function where applicable

cvinayak, 2020-10-06:
**Is your enhancement proposal related to a problem? Please describe.**
Refer to: https://github.com/zephyrproject-rtos/zephyr/pull/28932

**Describe the solution you'd like**
Refer to: https://github.com/zephyrproject-rtos/zephyr/pull/28932#issuecomment-704074630

**Describe alternatives you've considered**
NA

**Additional context**
NA

DhruvaG2000, 2021-09-17:
If I understand correctly, I need to move function definitions like ``void ull_drift_ticks_get()`` ([for eg](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/ull.c#L1892-L1936) )
and other places wherever there is ``#if defined(CONFIG_BT_PERIPHERAL)`` like these:
1. https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/ull.c#L205
2. https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/ull.c#L986

into [ull_slave.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/ull_slave.c) right?

DhruvaG2000, 2021-09-18:
Kindly assign this issue to me if my PR seems okay...
I will continue once I get assigned to make sure that I understand this issue and solution correctly

cvinayak, 2024-02-13:
Not planned in the near future.

closed 2024-02-13 by cvinayak