zephyrproject-rtos/zephyr #28962: Support SPI frame sizes larger than 64 bit

hendriklehmkuhl, 2020-10-06:
**Is your enhancement proposal related to a problem? Please describe.**
We would like to transmit SPI frames with a frame size of 480 byte to control an LED strip but the Zephyr SPI configuration struct (see spi.h) only supports a maximum frame size of 64 bit since the field word_size can only hold 6 bit. We are currently working with the NXP RT1064 evaluation board which SPI driver supports frame sizes of up to 512 byte. 

**Describe the solution you'd like**
Consideration of larger SPI frame sizes in spi.h

**Describe alternatives you've considered**
none

**Additional context**
The configuration function in the spi_mcux_lpspi.c driver even checks if the frame size is larger than 4096 bit to throw an error in that case but this can never be true.

lowlander, 2020-10-09:
The `word_size`  refers to the "word" that is send, most of the time 8 bit, but could in some special cases also be 16bit or 9 bit (but I am not even sure if all hardware supports "strange" (non 8bit)  word-sizes). 

But aren't you just wanting to send/recv 480 8bit "words" ? Like in the `samples/drivers/spi_fujitsu_fram/` where a 1024 byte buffer is written to the fram.

hendriklehmkuhl, 2020-10-30:
When using SPI to drive an LED strip (as for example in the ws2812b led strip driver sample) the SPI timings on the data transmission line are very important, since the LEDs are controlled by certain widths of high or low intervals (the clock line is not used here). When using 480 8bit frames I can observe "gaps" with my oscilloscope in between my 8-bit frames that lead to the LEDs misinterpreting the signal and showing a wrong colour. Therefore I would like to send 480 Byte in one frame without any gaps.

As i wrote above, the mcux spi driver does support such a behaviour but the zephyr API is not allowing me to use it.