zephyrproject-rtos/zephyr #28990: Docs: Dead links to sample source directories

martinjaeger, 2020-10-07:
Links to sample source files point to `docs.zephyrproject.org` instead of `github.com`.

Example: https://docs.zephyrproject.org/latest/samples/net/gsm_modem/README.html

Link behind `samples/net/gsm_modem` points to:

```https://docs.zephyrproject.org/blob/master/samples/net/gsm_modem```

instead of:

```https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/net/gsm_modem```

nashif, 2020-10-07:
@utzig you recently made a change here (#28245), can you please take a look?

utzig, 2020-10-07:
This seems to happen after the update to `west==0.8.0`; previously `WestManifest.from_file().get_projects(['zephyr'])[0].url` was `None`, now with `0.8.0` it is not anymore, so I'll check what changed there. Any tips @mbolivar-nordic?

mbolivar-nordic, 2020-10-07:
>  Any tips @mbolivar-nordic?

Looks like you figured it out from the API changes documentation?

PR #29004 (2020-10-07): doc: fix broken link-roles on west>=0.8
Fix broken links generated by link-roles after west was upgrade to 0.8.0. `url` was previously `None` and now returns an empty str.

Fixes #28990
closed 2020-10-07 by nashif