zephyrproject-rtos/zephyr #28992: Need for samples for Isochronous Channels feature

BOZHENG001, 2020-10-07:
**Is your feature request related to a problem? Please describe.**
I can see Zephyr 2.4.99 supporting the BLE new feature Isochronous Channels, and I also found the files and codes relating to it, but I cannot find any samples telling me how we can use it.

**Describe the solution you'd like**
It would be better if we can have some samples similar to "central_hr" and "peripheral_hr" telling us how we should set up the BLE devices for this Isochronous Channels feature.

govindadas, 2020-10-29:
> **Is your feature request related to a problem? Please describe.**
> I can see Zephyr 2.4.99 supporting the BLE new feature Isochronous Channels, and I also found the files and codes relating to > it, but I cannot find any samples telling me how we can use it.

Hi,
I could not find any LL code for Isochronous Channels in the sources. Yes, there is some host code for LE Audio (https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/bluetooth/host/audio/iso.c) but nothing on the controller part yet.

kartben, 2023-08-20:
Several ISO samples are now available
#32799

closed 2023-08-20 by kartben