zephyrproject-rtos/zephyr #28998: net: if: extend list of admin/operational states of network interface

mniestroj, 2020-10-07:
**Is your enhancement proposal related to a problem? Please describe.**
Currently there is a single flag (`NET_IF_UP`) that controls and notifies whether interface is ready (up) or not (down) for data transmission. It is set or cleared using `net_if_up()` and `net_if_down()` APIs. With current documentation it is not clear whether those functions are intended to be used by network drivers (e.g. to notify about ethernet link or WiFi being connected) or network administrative layer (e.g. application, connection manager, shell). At this point `net_if_up()` is mostly used network drivers to notify about them being ready, but there is also a shell command that calls that function. As a result it is confusing when can `net_if_up()` and `net_if_down()` be used.

**Describe the solution you'd like**
First of all we need clear documentation of `net_if_up()` and `net_if_down()` regarding what purpose do they serve, either administrative up/down (e.g. by application code) or to notify about network driver/carrier being ready or not.

For things like PPP we need to have administrative up/down commands, so PPP connection can be gracefull shut down. Calling `net_if_down()` immediately disables any communication over its net_if, including all control commands. This means that we need to add support for second pair of API.

We can adapt Linux approach described at https://www.kernel.org/doc/html/latest/networking/operstates.html, more specifically "Linux distinguishes between administrative and operational state of an interface.".

**Describe alternatives you've considered**
This is an open topic, so any ideas are welcome.

endian-benjamin, 2020-10-07:
Very good initiative.

jukkar, 2020-10-08:
> First of all we need clear documentation of `net_if_up()` and `net_if_down()` regarding what purpose do they serve, either administrative up/down (e.g. by application code)

Originally the `net_if_up()` and `net_if_down()` were used for administrative up/down from application code. Currently they are used in similar way from the driver which is probably not exactly we want here.
Using something like what is described in https://tools.ietf.org/html/rfc2863 sounds like a good plan.
```
/* RFC 2863 operational status */
enum {
	IF_OPER_UNKNOWN,
	IF_OPER_NOTPRESENT,
	IF_OPER_DOWN,
	IF_OPER_LOWERLAYERDOWN,
	IF_OPER_TESTING,
	IF_OPER_DORMANT,
	IF_OPER_UP,
};
```

AntoineZen, 2020-10-08:
> Currently they are used in similar way from the driver which is probably not exactly we want here.

Shouldn't the drivers use `net_eth_carrier_on()` and `net_eth_carrier_off()`?

jukkar, 2020-10-08:
> Shouldn't the drivers use `net_eth_carrier_on()` and `net_eth_carrier_off()`?

These are specific only to Ethernet.

eHammarstrom, 2020-10-12:
@jukkar how does the `IF_OPER_*` states relate to the net_mgmt `IF_UP/DOWN`, would that enum also be used by net_mgmt?

Should `net_if_oper_up/down/*` be introduced in the API namespace to separate application and driver usage?

jukkar, 2020-10-12:
> how does the `IF_OPER_*` states relate to the net_mgmt `IF_UP/DOWN`, would that enum also be used by net_mgmt?

We could introduce new events for these states, I see no issues with that.

> 
> Should `net_if_oper_up/down/*` be introduced in the API namespace to separate application and driver usage?

Yes, that sounds like a good idea.

rlubos, 2022-09-01:
I've investigated the issue a bit as it has been brought back in https://github.com/zephyrproject-rtos/zephyr/issues/49451.

I think the model proposed in the linked documents makes a lot of sense, it sounds reasonable to separate the administrative state from the operational state. It seems that it will require a lot of rework across the net stack/drivers but I guess it's unavoidable to sort out the situation.

A rough idea of what should be done, for discussion:
1. Treat currently present `NET_IF_UP` as an administrative state, thereby limiting the `net_if_up()`/`net_if_down()` use to application code only. 
2. As it is needed to differentiate whether the interface was only brought up by the application or is actually ready to transmit/receive packets, we'd need a new flag. For this purpose introduce a new `NET_IF_RUNNING` flag, controlled by the operational state. This, along with `NET_IF_UP` could be verified by the application whether the interface is ready to send application data. For example, if the application brings an Ethernet interface up with `net_if_up()` but the Ethernet cable is not plugged in, `NET_IF_UP` would be set but `NET_IF_RUNNING` not.
3. We'd likely also need new net_mgmt events to notify that interface is ready to use.

Now for the drivers/L2s to notify their state we'd need new APIs, to update interface's operational state:
1. A generic `net_if_carrier_up()`/`net_if_carrier_down()` to indicate whether interface is physically ready to transmit any packets. Currently, we only have Ethernet-specifc variant of those. I guess that for wireless intefraces (WiFi, OpenThread)  `net_if_carrier_up()` could be called as soon as radio driver is initialized, for wired intefraces, once cable is plugged in (if possible to verify, not sure about PPP).
2. For interfaces that need additional association (PPP, WiFi, OpenThread) the driver would get `net_if_dormant_on()`/`net_if_dormant_off()` to indicate whether the interface enters the association state. Once the interface is associated with the network, it'd call `net_if_dormant_off()`. As the association does not need to take place automatically, an interface could remain in the dormant state until for example an application code issue a WiFi connect command on an interface.

Based on `carrier` and `dormant` states the `NET_IF_RUNNING` flag would be set accordingly, i.e in the case of carrier up and dormant off, the interface would be considered ready to transmit application data.

I know this is nothing innovative, mostly summarizing information from https://www.kernel.org/doc/html/latest/networking/operstates.html, but it seems to be implementable in the Zephyr domain. Any thoughts?

krish2718, 2022-09-01:
LGTM, one query

> Treat currently present NET_IF_UP as an administrative state

Do we have a usecase for L2 driver is ready but the interface is disabled? If yes, we need both admin UP and oper UP, to distinguish its expected or some fault with L2 driver.

E.g., An application is built with multiple L2's but at run-time (need to extend `net iface` shell command) we can disable an interface (admin DOWN) to use preferred L2.

mniestroj, 2022-09-01:
>E.g., An application is built with multiple L2's but at run-time (need to extend net iface shell command) we can disable an interface (admin DOWN) to use preferred L2.

We actually have `net iface up` and `net iface down` shell commands now. Those would be the first users for ADMIN UP and ADMIN DOWN. Your application example is also valid, like for IoT devices that want to implement some high level logic on which network interface to use at the time. This might be also per end-user configuration, so either WiFi is conifgured or some other medium. Based on that one or the other interface would be put up/down.

jukkar, 2022-09-06:
> 3\. We'd likely also need new net_mgmt events to notify that interface is ready to use.

Hmm, at least for Ethernet we already have CARRIER ON/OFF event. From application point of view, it should not matter whether the interface is up/down because of administrative or operational status, so shouldn't it be enough if we have only one IFACE_UP/DOWN event?

krish2718, 2022-09-06:
>  From application point of view, it should not matter whether the interface is up/down because of administrative or operational status

Might still be helpful, as if its administratively down, then application can bring the interface up, but if its operationally down then it has to wait or bring it up by sending necessary commands.

jukkar, 2022-09-06:
> Might still be helpful, as if its administratively down, then application can bring the interface up, but if its operationally down then it has to wait or bring it up by sending necessary commands.

If the interface is oper down, admin cannot bring it up i.e., the command will fail.

rlubos, 2022-09-06:
> Hmm, at least for Ethernet we already have CARRIER ON/OFF event. From application point of view, it should not matter whether the interface is up/down because of administrative or operational [...]

Perhaps for the application, a single UP/DOWN even would be enough (so that UP is only issued if the interface is in an operational UP state). But on the other hand, I can imagine a module which for example is managing WiFi connections, for such a module a separate event when the interface was brought admin UP by the application could be useful so that it knows when to start WiFi association. And events like Ethernet Carrier ON/OFF I believe could be generalized.

> > Might still be helpful, as if its administratively down, then application can bring the interface up, but if its operationally down then it has to wait or bring it up by sending necessary commands.
>
> If the interface is oper down, admin cannot bring it up i.e., the command will fail.

I wonder, should such an operation really fail? For example, if an Ethernet cable was plugged out, the interface would remain in  operational DOWN state, but I think the application should still be able to call `net_if_up()` on such interface with a success? It'd change the admin state of the interface, but it would remain in oper DOWN state until the cable was plugged in (and such event would set the admin RUNNING flag on an interface).

jukkar, 2022-09-06:
> I wonder, should such an operation really fail? For example, if an Ethernet cable was plugged out, the interface would remain in operational DOWN state, but I think the application should still be [...]

Yes, I tried to say that but probably did not explain myself clarly (I meant that failure in this case would mean that interface stays down until "oper up" is entered). IMHO the "admin up" would be remembered and when "oper up" state is entered, the interface would become automatically up in that case. So we should honor the current "admin" state when "oper up" state is entered.

rlubos, 2022-09-06:
Some feedback from Networking Forum (September 2022):
* We should try to retain the meaning of `NET_EVENT_IF_UP`, to retain compatibility with existing applications. The event should indicate that the interface is up and running.
* Other states (carrier/dormant/admin up) could be reflected by new events if needed.

PR #51240 (2022-10-20): net: if: Improve interface status reporting
Extend interface state management as described in https://www.kernel.org/doc/html/latest/networking/operstates.html and https://www.rfc-editor.org/rfc/rfc2863.

With those changes, `net_if_up()`/`net_if_down()` are only supposed to be called by the application layer, not drivers/L2s. The existing codebase was refactored to reflect that.

Drivers/L2s have not two new flags to reflect their status - `NET_IF_LOWER_UP` and `NET_IF_DORMANT`, along with the helper functions. The combination of those two, along with the administrative state of the interface [... cut]
closed 2022-10-20 by carlescufi