zephyrproject-rtos/zephyr #28999: STM32: Transition to device tree based pinctrl configuration

erwango, 2020-10-07:
Following merge of https://github.com/zephyrproject-rtos/zephyr/pull/25996, STM32 boards should now be converted to the use of device tree pinctrl definitions.

This ticket aims at tracking the task completion:
| Drivers (PR) | Status | Boards | Status | Macros deprecation | Status |
| --- | --- | --- | --- | --- | --- |
| Serial (#25996)  | Merged  | #28954 |  |
| PWM (#28858) |   | #28858 |  |
| I2C () |   |  |
| SPI () |   |  |
| I2S () |   |  |
| ADC () |   |  |
| DAC () |   |  |
| Ethernet () |   |  |
| CAN () |   |  |
| USB () |   |  |

galak, 2020-10-07:
I'm also doing a test to see if we maybe split board changes into 3 groups (nucleo, stm32, other) w/regards to PRs.

galak, 2020-10-07:
@erwango / @lochej / @gmarull  I've worked up a python script to convert pinmux.c into .dts.  Seems to work for a good number of cases (I2C, SPI, ADC, DAC, I2S).  There are some corner cases that need some discussion / understanding.

gmarull, 2020-10-07:
It would be great if you can share the script @galak . Once all boards get the right pinctrl included (I guess it will be the case after uart PR) it will help a lot. I can take a couple more drivers, it’s rather quick to add support for pinctrl, so together with the script we can finally get rid of pinmux in a few days :-)

galak, 2020-10-08:
> It would be great if you can share the script @galak . Once all boards get the right pinctrl included (I guess it will be the case after uart PR) it will help a lot. I can take a couple more [...]

Happy to share, just need to figure out best way to do that:

Here are some areas that we need to look into:

I2C: [I2C2 missing]
nucleo_f030r8
stm32vl_disco
olimexino_stm32
stm32_min_dev_black
stm32_min_dev_blue
olimex_stm32_h103
waveshare_open103z

CAN:
waveshare_open103z [can missing]
olimexino_stm32 [can missing]
stm32f3_disco [can missing]
stm32f072b_disco [can missing]
nucleo_l452re [can/can1]
nucleo_l452re_p [can/can1]

USB:
disco_l475_iot1 - missing usb_otg_fs_id_pa10
b_l4s5i_iot01a - missing usb_otg_fs_id_pa10
nucleo_l4r5zi - missing usb_otg_fs_id_pa10
waveshare_open103z - usb_dm_pa11 missing
nucleo_wb55rg - usb_dm_pa11 missing

galak, 2020-10-08:
Here the current script:

https://github.com/galak/zephyr/tree/st-pin-convert
https://raw.githubusercontent.com/galak/zephyr/695016205558bd3b1c2711b46bf5af842ccb5a2b/boards/arm/st-pin-convert.py

Its kinda in a hackish state and currently line 29 can be hacked to enable different driver classes (mac/sdmmc1 don't have any pins define in the dtsi files right now so they don't do anything).

martinjaeger, 2020-10-08:
Just implemented the DAC part and updated the table. I think we can summarize all board updates into a single commit, as it's only a few lines of code changed in most cases.

I can look into the CAN driver later (or tomorrow) as I'm working on that anyway at the moment.

galak, 2020-10-08:
> Just implemented the DAC part and updated the table. I think we can summarize all board updates into a single commit, as it's only a few lines of code changed in most cases.
> 
> I can look into the CAN driver later (or tomorrow) as I'm working on that anyway at the moment.

@martinjaeger for CAN be aware of https://github.com/zephyrproject-rtos/hal_stm32/pull/70

erwango, 2020-10-08:
@galak, @gmarull, @lochej, @martinjaeger 
Please note important change done in API in #29057

Nukersson, 2020-10-13:
@gmarull 

Just tested ETH pinctrl for STM32 as follows (add `modules/hal/stm32/scripts/genpinctrl/stm32-pinctrl-config.yaml`):
```
- name: ETH
  match: "^ETH_*"
  mode: alternate
  bias: disable
```

Could we give it a try?

Just for info: For `h7`series all the pins have `very high speed`, `no push-pull` and `AF11` (see `zephyr/drivers/pinmux/stm32/pinmux_stm32h7.h`)

closed 2020-11-03 by carlescufi