zephyrproject-rtos/zephyr #29007: OOT manifest+module discovery/builds fail

gregshue, 2020-10-07:
**Describe the bug**
Building a project from a manifest+module topology (T2) now fails within zephyr/scripts/zephyr_module.py with `TypeError: expected str, bytes or os.PathLike object, not NoneType`

**To Reproduce**
Steps to reproduce the behavior:
1.` west init -m https://github.com/gregshue/oot_module_example --mr main --mf zephyr/manifests/zp/zephyr-v2.4.0/west.yml tmp && cd tmp && west update && source zephyr/zephyr-env.sh`
2. `west build -b native_posix zephyr/samples/hello_world/`
3. See error

**Expected behavior**
I expect this to successfully build, as available on zephyr-v2.3.0.

**Impact**
This is a showstopper, preventing me from using Zephyr beyond v2.3.0 without branching and creating downstream solutions.

**Logs and console output**
zephyr@zephyr-VirtualBox:~/git/tmp$ west build -b native_posix zephyr/samples/hello_world/
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/zephyr/git/tmp/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/zephyr/git/tmp/zephyr/samples/hello_world
-- Zephyr version: 2.4.0 (/home/zephyr/git/tmp/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components: Interpreter
-- Found west (found suitable version "0.8.0a1", minimum required is "0.7.1")
CMake Error at /home/zephyr/git/tmp/zephyr/cmake/zephyr_module.cmake:48 (message):
  Traceback (most recent call last):

    File "/home/zephyr/git/tmp/zephyr/scripts/zephyr_module.py", line 326, in <module>
      main()
    File "/home/zephyr/git/tmp/zephyr/scripts/zephyr_module.py", line 266, in main
      meta = process_module(project)
    File "/home/zephyr/git/tmp/zephyr/scripts/zephyr_module.py", line 99, in process_module
      module_path = PurePath(module)
    File "/usr/lib/python3.6/pathlib.py", line 624, in __new__
      return cls._from_parts(args)
    File "/usr/lib/python3.6/pathlib.py", line 656, in _from_parts
      drv, root, parts = self._parse_args(args)
    File "/usr/lib/python3.6/pathlib.py", line 640, in _parse_args
      a = os.fspath(a)

  TypeError: expected str, bytes or os.PathLike object, not NoneType

Call Stack (most recent call first):
  /home/zephyr/git/tmp/zephyr/cmake/app/boilerplate.cmake:150 (include)
  /home/zephyr/git/tmp/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/zephyr/git/tmp/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/zephyr/git/tmp/build -S/home/zephyr/git/tmp/zephyr/samples/hello_world -GNinja -DBOARD=native_posix
zephyr@zephyr-VirtualBox:~/git/tmp$

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: zephyr-v2.4.0

**Additional context**

nashif, 2020-10-07:
@tejlmand  @mbolivar-nordic FYI

mbolivar-nordic, 2020-10-07:
https://github.com/gregshue/oot_module_example/blob/main/zephyr/manifests/zp/zephyr-v2.4.0/west.yml#L41

This is invalid manifest syntax. See https://github.com/zephyrproject-rtos/zephyr/issues/28925 for details.

closed 2020-10-07 by mbolivar-nordic

tejlmand, 2020-10-08:
> https://github.com/gregshue/oot_module_example/blob/main/zephyr/manifests/zp/zephyr-v2.4.0/west.yml#L41
> 
> This is invalid manifest syntax. See #28925 for details.

@mbolivar-nordic please paste a permalink next time.
Looks like code has been forced pushed (at least I see no errors).

Should we make a better error handling here ?
Seems wrong to me that users can successfully do `west init` and `west update` when using a `west.yml` with invalid syntax.
Makes it very hard to understand that it's a manifest error when they see a stacktrace when running CMake.

mbolivar-nordic, 2020-10-09:
> @mbolivar-nordic please paste a permalink next time.
> Looks like code has been forced pushed (at least I see no errors).

It was an empty `self: path:`.

> Should we make a better error handling here ?

We already did as a follow up to https://github.com/zephyrproject-rtos/zephyr/issues/28925.

The west commit is https://github.com/zephyrproject-rtos/west/commit/da7668699aaf9b3a6ed416f092a0d02c0958f2ce.

tejlmand, 2020-10-09:
> The west commit is [zephyrproject-rtos/west@da76686](https://github.com/zephyrproject-rtos/west/commit/da7668699aaf9b3a6ed416f092a0d02c0958f2ce).

Oh, right.
Was testing using `v0.8.0` and not paying attention to the new west PR that I actually approved.