zephyrproject-rtos/zephyr #29013: [Coverity CID :214870] Untrusted loop bound in drivers/ethernet/eth_w5500.c

zephyrbot, 2020-10-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/ethernet/eth_w5500.c

Category: Insecure data handling
Function: `w5500_rx`
Component: Drivers
CID: [214870](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214870)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

parthitce, 2020-10-23:
I couldn't access the report for this coverity. How to view the scan report/log?

jukkar, 2020-10-23:
```
static void w5500_rx(const struct device *dev)
{
        uint8_t mask = 0;
        uint8_t header[2];
        uint8_t tmp[2];
        uint16_t off;
        uint16_t rx_len;
        uint16_t rx_buf_len;
        uint16_t read_len;
        uint16_t reader;
        struct net_buf *pkt_buf = NULL;
        struct net_pkt *pkt;
        struct w5500_runtime *ctx = dev->data;
        const struct w5500_config *config = dev->config;
        /* disable interrupt */
        w5500_spi_write(dev, W5500_SIMR, &mask, 1);
        w5500_spi_read(dev, W5500_S0_RX_RSR, tmp, 2);
        rx_buf_len = sys_get_be16(tmp);
1. Condition rx_buf_len == 0, taking false branch.
        if (rx_buf_len == 0) {
                return;
        }
        w5500_spi_read(dev, W5500_S0_RX_RD, tmp, 2);
        off = sys_get_be16(tmp);
        w5500_readbuf(dev, off, header, 2);
2. tainted_data_return: Function sys_get_be16 returns tainted data. [show details]
3. lower_bounds: Casting narrower unsigned sys_get_be16(header) to wider signed type int effectively tests its lower bound.
4. var_assign_var: Assigning: rx_len = sys_get_be16(header) - 2. Both are now tainted.
        rx_len = sys_get_be16(header) - 2;
        pkt = net_pkt_rx_alloc_with_buffer(ctx->iface, rx_len,
                        AF_UNSPEC, 0, K_MSEC(config->timeout));
5. Condition !pkt, taking false branch.
        if (!pkt) {
                eth_stats_update_errors_rx(ctx->iface);
                return;
        }
        pkt_buf = pkt->buffer;
6. var_assign_var: Assigning: read_len = rx_len. Both are now tainted.
        read_len = rx_len;
        reader = off + 2;
        do {
                size_t frag_len;
                uint8_t *data_ptr;
                size_t frame_len;
                data_ptr = pkt_buf->data;
                frag_len = net_buf_tailroom(pkt_buf);
                if (read_len > frag_len) {
                        frame_len = frag_len;
                } else {
                        frame_len = read_len;
                }
                w5500_readbuf(dev, reader, data_ptr, frame_len);
                net_buf_add(pkt_buf, frame_len);
                reader += frame_len;
                read_len -= frame_len;
                pkt_buf = pkt_buf->frags;
CID 214870 (#1 of 1): Untrusted loop bound (TAINTED_SCALAR)7. tainted_data: Using tainted variable read_len as a loop boundary.
        } while (read_len > 0);
        if (net_recv_data(ctx->iface, pkt) < 0) {
                net_pkt_unref(pkt);
        }
        sys_put_be16(off + 2 + rx_len, tmp);
        w5500_spi_write(dev, W5500_S0_RX_RD, tmp, 2);
        w5500_command(dev, S0_CR_RECV);
}
```

carlescufi, 2021-02-09:
@parthitce do you have plans to address this?

parthitce, 2021-02-09:
> @parthitce do you have plans to address this?

@carlescufi Yes I will be addressing this along with a functional bug in w5500 driver before end of this month. Was busy with usual business.

galak, 2021-05-11:
@parthitce any updates on this?

galak, 2021-06-01:
@parthitce  ping

parthitce, 2021-06-01:
@galak Sorry I didn't find time checking it in past month. Will do it in the following week. Also W5500 driver is not functional or regression from networking changes. Will sort it out.

marcel-cd, 2021-08-12:
@parthitce  Can you please have a look at the following two changes we did to get the W5500 driver running?

[eth_w5500_diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/6973356/eth_w5500_diff.txt)

marcel-cd, 2021-08-12:
@parthitce Can you have a look at the changes I did to get the W5500 running?

[eth_w5500_diff.txt](https://github.com/zephyrproject-rtos/zephyr/files/6973377/eth_w5500_diff.txt)

parthitce, 2021-08-12:
@marcel-cd Thanks, this looks fine. Please open it as PR and I can test it.

marcel-cd, 2021-08-12:
@parthitce ok, done: https://github.com/zephyrproject-rtos/zephyr/pull/37648
I hope everything is fine, my first PR to the Zephyr Project.

closed 2022-01-04 by github-actions[bot]