zephyrproject-rtos/zephyr #29014: [Coverity CID :214872] Bad bit shift operation in drivers/ethernet/eth_w5500.c

zephyrbot, 2020-10-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/drivers/ethernet/eth_w5500.c

Category: Integer handling issues
Function: `w5500_set_config`
Component: Drivers
CID: [214872](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214872)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

nashif, 2020-10-07:
@parthitce FYI

nashif, 2020-10-07:
```

338static int w5500_set_config(const struct device *dev,
339                            enum ethernet_config_type type,
340                            const struct ethernet_config *config)
341{
342        uint8_t mode;
   1. assignment: Assigning: mr = 128.
343        uint8_t mr = W5500_S0_MR_MF;
344
345        w5500_spi_read(dev, W5500_S0_MR, &mode, 1);
   2. Condition 0, taking false branch.
346        if (IS_ENABLED(CONFIG_NET_PROMISCUOUS_MODE) &&
347            type == ETHERNET_CONFIG_TYPE_PROMISC_MODE) {
348                if (config->promisc_mode) {
349                        if (!(mode & W5500_S0_MR_MF))
350                                return -EALREADY;
351                        }
352
353                        /* clear */
354                        WRITE_BIT(mode, mr, 0);
355        } else {
   3. Condition mode & mr, taking false branch.
356                if (mode & mr) {
357                        return -EALREADY;
358                }
359
360                /* set */
   4. Condition 1, taking true branch.
   CID 214872 (#1 of 1): Bad bit shift operation (BAD_SHIFT)5. large_shift: In expression 1UL << mr, left shifting by more than 31 bits has undefined behavior. The shift amount, mr, is 128.
361                WRITE_BIT(mode, mr, 1);
362        }
363
364        return w5500_spi_write(dev, W5500_S0_MR, &mode, 1);
365}
```

closed 2020-10-26 by nashif