zephyrproject-rtos/zephyr #29018: [Coverity CID :214881] Unchecked return value in subsys/mgmt/ec_host_cmd/ec_host_cmd_handler.c

zephyrbot, 2020-10-07:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/master/subsys/mgmt/ec_host_cmd/ec_host_cmd_handler.c

Category: Error handling issues
Function: `handle_host_cmds_entry`
Component: Other
CID: [214881](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=214881)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

jenmwms, 2020-10-15:
Hi @laurenmurphyx64  I see you submitted a fix. Thanks! I'll review soon. When do you plan to update Coverity issue? Were you wanting to wait until the fix is merged to update https://scan9.coverity.com/reports.htm#v39290/p12996/fileInstanceId=71339256&defectInstanceId=8259379&mergedDefectId=214881?

laurenmurphyx64, 2020-10-15:
@jenmwms Thanks for the reminder! Yes, I'll wait until it's approved and merged if that's alright

PR #29207 (2020-10-21): ec_host_cmd: Fix unused return value from k_sem_take call
Fixes Coverity issue 214881.

Fixes #29018
commit b82efb34 (2020-10-21): ec_host_cmd: Fix unused return value from k_sem_take call

Fixes Coverity issue 214881.

Fixes #29018
closed 2020-10-21 by galak