zephyrproject-rtos/zephyr #29059: HAL: mchp: Missing PCR ids to control PM for certain HW blocks

albertofloyd, 2020-10-08:
**Describe the bug**

While there is control power management for different each HW instance  of certain technology inside HAL (pcr.h)
i.e. 	
        PCR_PS2_0 = (384u + 5u),
	PCR_PS2_1 = (384u + 6u),

	PCR_PWM1 = (128u + 20u),
	PCR_PWM2 = (128u + 21u),
	PCR_PWM3 = (128u + 22u),
	PCR_PWM4 = (128u + 23u),
	PCR_PWM5 = (128u + 24u),
	PCR_PWM6 = (128u + 25u),
	PCR_PWM7 = (128u + 26u),
	PCR_PWM8 = (128u + 27u),

I2C and SMB are only exposing 3 ids, need the location the missing 2 in each case.
	PCR_I2C0 = (512u + 10u),
	PCR_I2C1 = (512u + 11u),
	PCR_I2C2 = (512u + 12u),

Also TACH there is only 1 PCR id.

scottwcpg, 2020-10-09:
MEC15xx has two groups of I2C controllers:
5 controllers named SMB0, SMB1, ..., SMB4 that include the network layer hardware on top of the base I2C.
3 controllers named I2C0, I2C1, ..., I2C2 that do not include the network layer hardware.  These controller have most of the same register set except for registers related to the network layer hardware.
This is the reason for the different naming convention.
The pcr.h header has:
PCR_SMB0, PCR_SMB1, PCR_SMB2, PCR_SMB3, PCR_SMB4
PCR_I2C0, PCR_I2C1, and PCR_I2C 
These defines encode the register PCR register offset and bit position.
The header also has:
MCHP_PCR1_SMB0_POS	10u
MCHP_PCR3_SMB1_POS	13u
MCHP_PCR3_SMB2_POS	14u
MCHP_PCR3_SMB3_POS	15u
MCHP_PCR3_SMB4_POS	20u
MCHP_PCR4_I2C0_POS	10u
MCHP_PCR4_I2C1_POS	11u
MCHP_PCR4_I2C3_POS	12u

MCHP_PCR1_TACH0_POS	2u
MCHP_PCR1_TACH1_POS	11u
MCHP_PCR1_TACH2_POS	12u
MCHP_PCR1_TACH3_POS	13u
PCR_TACH0 = (128u + 2u),
PCR_TACH1 = (128u + 11u),
PCR_TACH2 = (128u + 12u),
PCR_TACH3 = (128u + 13u),

Is this a request to update C code in the soc and board layer to program these bits?

albertofloyd, 2020-10-13:
@scottwcpg  clarification above is enough

closed 2020-10-13 by albertofloyd