zephyrproject-rtos/zephyr #29066: kernel: k_sleep doesn't handle relative or absolute timeouts >INT_MAX

JordanYates, 2020-10-09:
**Describe the bug**

The primary sleep function `z_tick_sleep` in `kernel/sched.c` does not respect the type of `k_ticks_t`, instead using `int32_t`.
There are two main consequences of this:

1. Sleeps of tick duration greater than `INT_MAX` will not behave properly.
2. Absolute timeouts fail after `INT_MAX` ticks (18 hours on nRF).

In the second case, timout.ticks will have some value less than `INT_MIN`. This becomes a postive number near `INT_MAX` after being cast to `int32_t`. The `Z_TICK_ABS` check in `z_add_timeout` fails, resulting in a sleep of approximately 18hrs being scheduled, instead of ~1 second.

**To Reproduce**

Initialize `curr_tick` in `kernel\timeout.c` to just before rollover:
```
static uint64_t curr_tick = INT32_MAX - 320000;
``` 
Modify the main loop of `samples\basic\blinky` to the following:
```
	int64_t timeout = k_uptime_get() + 1000;

	while (1) {
		k_sleep(K_TIMEOUT_ABS_MS(timeout));
		timeout += 1000;

		gpio_pin_set(dev, PIN, (int)led_is_on);
		led_is_on = !led_is_on;
	}
```
On NRF boards, the LED will stop blinking after ~10 seconds.

**Expected behavior**
Absolute timeouts (and large relative timeouts) should delay for the specified duration.

**Impact**
High priority/showstopper, I use absolute timeouts to mitigate drift in our main loop.
This bug means the watchdog is rebooting my application every 18 hours.

**Logs and console output**
By adding `printk("Ticks: %lld %lld\n", ticks, Z_TICK_ABS(ticks));` to `z_add_timeout`
```
Ticks: -2147324823 2147324821
Ticks: -2147357591 2147357589
Ticks: -2147390359 2147390357
Ticks: -2147423127 2147423125
Ticks: -2147455895 2147455893
Ticks: 2147478633 -2147478635
```

**Environment (please complete the following information):**
- Zephyr 2.4

andyross, 2020-10-09:
Indeed, looks like z_tick_sleep() takes an int32_t still and not a k_ticks_t.  Not sure why this doesn't produce a size warning when called and k_ticks_t is a 64 bit type...

Note that for this to work at all, CONFIG_TIMEOUT_64BIT needs to be set.

JordanYates, 2020-10-12:
Adding `#pragma GCC diagnostic error "-Wconversion"` to the top of `sched.c` does produce (among a multitude of others):
```
C:/Users/Jordan/code/project/zephyr/kernel/sched.c: In function 'z_impl_k_sleep':
C:/Users/Jordan/code/project/zephyr/kernel/sched.c:1265:23: error: conversion from 'k_ticks_t' {aka 'long long int'} to 'int32_t' {aka 'int'} may change value [-Werror=conversion]
 1265 |  ticks = z_tick_sleep(ticks);
```
Based on `compiler_flags.cmake`, this option is only enabled for warning levels 2 and 3.
Unfortunately there are so many warnings at these levels that they are functionally useless.

If the build system supported it, I feel like it would be useful for the core kernel functionality to be compiled with the highest possible warning levels, even if device drivers etc aren't. That would help pick up on bugs like this one.

PR #29343 (2020-12-05): kernel: sched: Use k_ticks_t in z_tick_sleep
z_tick_sleep was using int32_t what could cause a possible overflow
when converting from k_ticks_t.

Fixes #29066
closed 2020-12-05 by nashif