zephyrproject-rtos/zephyr #29074: #27901 breaks mikroe_* shields overlay

agansari, 2020-10-09:
The change in order #27901 will put the cart before the horse when building with overlays.

**To Reproduce**

> west build -b lpcxpresso55s69_cpu0 samples/net/sockets/echo_server --pristine -- -DSHIELD=mikroe_eth_click

**Expected behavior**
Should build.

**Impact**
Does not build.

> Error: lpcxpresso55s69_cpu0.dts.pre.tmp:447.15-25 Label or path eth_click not found
> FATAL ERROR: Syntax error parsing input tree
> CMake Error at /mnt/c/1.Code/zephyrproject/zephyr/cmake/dts.cmake:211 (message):
>   command failed with return code: 1

Excerpt from temporary dts file (lpcxpresso55s69_cpu0.dts.pre.tmp):
```
/delete-node/ &eth_click; <---- board's overlay
&mikrobus_spi {
 status = "okay";
 eth_click: eth_click@1 {
  compatible = "microchip,enc28j60";
  reg = <0x1>;
  local-mac-address = [00 00 00 01 02 03];
  spi-max-frequency = <10000000>;
  int-gpios = <&mikrobus_header 7 (1 << 0)>;
  label = "ETH_CLICK";
 };
};
&mikrobus_spi { <---- shield's overlay
 status = "okay";
 eth_click: eth_click@0 {
  compatible = "microchip,enc28j60";
  reg = <0x0>;
  local-mac-address = [00 00 00 01 02 03];
  spi-max-frequency = <10000000>;
  int-gpios = <&mikrobus_header 7 (1 << 0)>;
  label = "ETH_CLICK";
 };
};
```

**Possible resolution**
Change the overlay include order or change the board overlay

tejlmand, 2020-10-26:
seems we need to define a rule on processing order, but let's first decide what the correct / expected behavior should be seen from the user perspective. And then from there determine the processing order in the build system.

When looking at:
https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/shields/mikroe_eth_click/mikroe_eth_click.overlay
and
https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/shields/mikroe_eth_click/boards/lpcxpresso55s69_cpu0.overlay

then I would actually expect that what is specified in `boards/lpcxpresso55s69_cpu0.overlay` takes precedence over `mikroe_eth_click.overlay`, so that a board overlay can overrule anything the shield default defines.

With this logic, the patch in: https://github.com/zephyrproject-rtos/zephyr/pull/27901 must be reverted, as change processing order seems to result in the definitions in the shield is taking precedence over the board.

@mbolivar-nordic wouldn't you agree ?

mbolivar-nordic, 2020-10-28:
> @mbolivar-nordic wouldn't you agree ?

I don't use shields, so I think this is an @erwango question.

erwango, 2020-10-28:
> With this logic, the patch in: #27901 must be reverted, as change processing order seems to result in the definitions in the shield is taking precedence over the board.

Fully agree.
#27901 shouldn't have been merged as this breaks overlay logic.

nashif, 2020-10-29:
> #27901 shouldn't have been merged as this breaks overlay logic.

what is the course of action here? Should this be reverted or do we have a way to fix this? @erwango  can you please propose a solution?

tejlmand, 2020-11-05:
Created: #29825 as followup to minimize the risk that this happens again.

PR #29824 (2020-11-10): Revert "shields: Modify overlay file order"
Fixes: #29074

This reverts commit fc8f639b9a6df12a024b0ee5b452db06d7b72609.

The suggestion provided in #27901 is impacting processing order of
overlay files in a non-logical way, see #29074 discussion for details.
commit 59bbc5a3 (2020-11-10): Revert "shields: Modify overlay file order"

Fixes: #29074

This reverts commit fc8f639b9a6df12a024b0ee5b452db06d7b72609.

The suggestion provided in #27901 is impacting processing order of
overlay files in a non-logical way, see #29074 discussion for details.
closed 2020-11-10 by nashif