zephyrproject-rtos/zephyr #29078: nRF52840 doesn't start legacy advertisment after extended advertisment

Lukas1811, 2020-10-09:
**Describe the bug**
Im currently developing an application on a nRF52840-DK, which should switch from normal advertisment to extended advertisment if a specific opcode is written to an characteristic. All of this works but if I want to switch back to normal advertisment, the controller won't advertise and return error -12 (ENOMEM) when calling _bt_le_adv_start()_

**To Reproduce**
Steps to reproduce the behavior:
1. Start extended advertisment using _bt_le_ext_adv_start()_
2. Stop extended advertisment using _bt_le_ext_adv_stop()_
3. Start legacy advertisment using _bt_le_adv_start()_

**Expected behavior**
The controller should switch back from extended advertisment to legacy advertisment

**Impact**
I have to reset the controller everytime to get it back to work properly.

**Environment:**
 - Zephyr 2.3.99
- nRF52840-DK (PCA10056)

**Additional context**
If I get it right, error -12 means ENOMEM, which is returned if _"No free connection objects available for connectable advertiser."_ ([Documentation](https://docs.zephyrproject.org/latest/reference/bluetooth/gap.html#c.bt_le_adv_start)). Thats why I set _CONFIG_BT_MAX_CONN=8_ in my config file but that changed nothing.

Lukas1811, 2020-10-09:
I found a solution for my problem, this problem only occurs if you don't delete the advertisment set before starting the legacy advertisment. I'm not sure if this should work like this or if this is a bug

joerchan, 2020-10-10:
That is how it works, with one advertising set. Otherwise the number of advertising sets must be increased (if supported by the controller)

closed 2020-10-10 by joerchan

duckgreenpepperjudo, 2023-07-19:
I have a similar problem. How do you delete an advertising set?