zephyrproject-rtos/zephyr #29083: Zephyr host gatt discover response missing permission value

boachair, 2020-10-09:
**Describe the bug**
After calling Zephyr API, bt_gatt_discover(), GATT client expects a response structured as Attribute, which contains one permission field (see below). However, it is always 0 now. It might be caused because the GATT server response does not update this attribute permission.

--
struct bt_gatt_attr {
/** Attribute UUID */
const struct bt_uuid *uuid;
...
/** Attribute user data */
void *user_data;
/** Attribute handle */
u16_t handle;
/** Attribute permissions */
u8_t perm;
};

**To Reproduce**
Steps to reproduce the behavior:
1. call Zephyr API, bt_gatt_discover() with discover type as the primary service
2. Read the permission field in the response (structured as Attribute) in the callback

**Expected behavior**
The Attribute Permissions shall be read-only (BT_GATT_PERM_READ).

**Impact**
annoyance

**Logs and console output**
None

**Environment (please complete the following information):**
- Zephyr 2.4.0
- nRF52840 DK

**Additional context**
None

joerchan, 2020-10-12:
@boachair What is the annoyance there? The permissions on the field is not sent, which is why it is not set of the attribute structure. I don't see setting it to "READ" as something that provides any useful information.

boachair, 2020-10-13:
There exists a difference in discovering service and characteristics, i.e., Characteristic discovery returns the permission field as 1 (READ only) in the callback, while Primary Service/included service discovery return this permission as 0. However, the requirement for attribute permission are the same for both.

joerchan, 2020-10-13:
@boachair Yes I noticed that, and I'd like to fix that. However what I would like to do is to set it to 0 in all cases, and then document that this field is not valid in the discovery callback.

Setting them to be "READ" would only be misleading, since we don't actually know the correct permissions, and there is no way to retrieve them either.

joerchan, 2020-10-14:
@nashif I set it to low instead. It's just an inconsistency in the parameters, nothing is missing.

PR #29163 (2020-10-15): Bluetooth: GATT: Consistent permission handling of discovered attribute
Be consistent in the permission handling of the discovered attribute
in the temporary object given in the discovery callback.
For characteristics the permission field was set to READ, while for
all other attributes it was set to 0.
    
Fixes: #29083

Document which fields in the attribute parameter in the discovery
callback contains valid information.
Document the type of the user_data so that the application can cast
this to a valid type in order to retrieve the information.
commit ee81c086 (2020-10-15): Bluetooth: GATT: Consistent permission handling of discovered attribute

Be consistent in the permission handling of the discovered attribute
in the temporary object given in the discovery callback.
For characteristics the permission field was set to READ, while for
all other attributes it was set to 0.

Fixes: #29083
closed 2020-10-15 by carlescufi