zephyrproject-rtos/zephyr #29087: Moving (some) boards to their own repo/module

lowlander, 2020-10-09:
The Zephyr `boards/*/` dirs are growing and growing (I have two bsp's in the pipeline myself). Would it be an idea to make a `zephyr_boards` repo/module where we collect all boards, and only have a very limited number of boards in the zephyr main repo ?

@martinjaeger mentioned that most board doku have an image, that kind of bloats the main zephyr repo. But he also pointed out, quote; _But the disadvantage is that it's much more difficult to keep them synchronized with the rest of the Zephyr repo if they are out of tree..._

If splitting of a `zephyr_boards` repo has more cons than  pros, maybe it would be an idea to at least make vendor subdirectories? like `boards/arm/st/nucleo*`, `boards/arm/nxp/*`, `boards/arm/olimex/*`, etc. ? (probably would look nicer in the docu also, instead of that endless list of boards)

Or maybe we can do a something like gstreamer, where they have good/ugly/bad repos (not necessarily with those names) where we have a `zephyr_boards` module/repo with boards that for example, not yet have complete documentation, or are not yet 100% complete (for example when the board has a CAN bus but the dts/pinmux doesn't implement it yet, but the parts that are there are working). And the boards in the zephyr main repo just have a higher quality and are fully documented.

nashif, 2022-11-17:
related issue to structure: https://github.com/zephyrproject-rtos/zephyr/issues/51831

nashif, 2022-11-17:
re image size: https://github.com/zephyrproject-rtos/zephyr/issues/49026

nashif, 2022-11-17:
re good/bad/ugly: https://github.com/zephyrproject-rtos/zephyr/issues/38566

nashif, 2022-11-17:
but, I do not think it is planned or feasible right now to split boards out, maybe after all of the above has been done and we went through some real cleanup and classification.

closed 2022-11-17 by nashif