zephyrproject-rtos/zephyr #29095: zefi.py has incorrect assertions

andrewboie, 2020-10-09:
**Describe the bug**
I ran into this build error:

```
[1/1] Linking C executable zephyr/zephyr.elf
FAILED: zephyr/zephyr.elf zephyr/zephyr.bin zephyr/zephyr.lst zephyr/zephyr.stat zephyr/zephyr.strip
: && ccache /home/apboie/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc -m64   zephyr/CMakeFiles/zephyr_final.dir/misc/empty_file.c.obj  -o zephyr/zephyr.elf  zephyr/arch/arch/x86/pagetables.o  zephyr/kobject_hash_renamed.o  -Wl,-T  zephyr/linker_pass_final.cmd  -Wl,-Map=/home/apboie/projects/zephyr2/zephyr/tests/kernel/mem_protect/mem_protect/build/zephyr/zephyr_final.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/x86/core/libarch__x86__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/x86/apollo_lake/libsoc__x86__apollo_lake.a  zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a  zephyr/drivers/serial/libdrivers__serial.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/arch/x86/core/offsets/offsets.c.obj  -L"/home/apboie/zephyr-sdk/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0"  -L/home/apboie/projects/zephyr2/zephyr/tests/kernel/mem_protect/mem_protect/build/zephyr  -lgcc  -m64  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /home/apboie/projects/zephyr2/zephyr/tests/kernel/mem_protect/mem_protect/build/zephyr && /usr/bin/cmake -E rename zephyr_final.map zephyr.map && /home/apboie/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-objcopy --gap-fill 0xff --output-target=binary --remove-section=.comment --remove-section=COMMON --remove-section=.eh_frame zephyr.elf zephyr.bin && /home/apboie/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-objdump -d -S zephyr.elf >zephyr.lst && /home/apboie/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-readelf -e zephyr.elf > zephyr.stat && /home/apboie/zephyr-sdk/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-strip --strip-all zephyr.elf -ozephyr.strip && /usr/bin/python3.8 /home/apboie/projects/zephyr2/zephyr/arch/x86/zefi/zefi.py -f /home/apboie/projects/zephyr2/zephyr/tests/kernel/mem_protect/mem_protect/build/zephyr/zephyr.elf
Traceback (most recent call last):
  File "/home/apboie/projects/zephyr2/zephyr/arch/x86/zefi/zefi.py", line 161, in <module>
    build_elf(args.elf_file)
  File "/home/apboie/projects/zephyr2/zephyr/arch/x86/zefi/zefi.py", line 55, in build_elf
    assert (h.p_filesz % 4) == 0
AssertionError
ninja: build stopped: subcommand failed.
```

This assertion is not correct. From dumping the file we have an unaligned filesiz/memsiz, I have no reason to believe our ELF binary is incorrect:
```
$ readelf -l zephyr/zephyr.elf

Elf file type is EXEC (Executable file)
Entry point 0x104b
There are 3 program headers, starting at offset 64

Program Headers:
  Type           Offset             VirtAddr           PhysAddr
                 FileSiz            MemSiz              Flags  Align
  LOAD           0x0000000000000100 0x0000000000001000 0x0000000000001000
                 0x0000000000006000 0x0000000000006000  RWE    0x20
  LOAD           0x0000000000007000 0x0000000000100000 0x0000000000100000
                 0x000000000001a000 0x00000000044af000  RWE    0x1000
  LOAD           0x0000000000021000 0x00000000045af000 0x00000000045af000
                 0x0000000000405b21 0x0000000000405b21  RW     0x20

 Section to Segment mapping:
  Segment Sections...
   00     .locore
   01     text initlevel app_shmem_regions log_const_sections rodata app_smem bss noinit
   02     datas devices _static_thread_data_area k_timer_area k_heap_area k_mutex_area k_msgq_area k_sem_area pagetables kobject_data
```

**To Reproduce**
Currently I can reproduce this with this PR https://github.com/zephyrproject-rtos/zephyr/pull/28986 and building tests/kernel/mem_protect/mem_protect on up_squared

**Expected behavior**
Build to complete successfully.

**Impact**
Showstopper

andrewboie, 2020-10-09:
This seems to work around the issue by ensuring the very last section is 4-byte aligned, but I have doubts that this is really the right way to address it:
```
--- a/include/linker/kobject.ld
+++ b/include/linker/kobject.ld
@@ -38,6 +38,7 @@
 #ifndef CONFIG_XIP
        *(".kobject_data.rodata*")
 #endif
+        . = ALIGN(4);
         } GROUP_DATA_LINK_IN(RAMABLE_REGION, ROMABLE_REGION)

 #ifdef CONFIG_GEN_PRIV_STACKS
```

andrewboie, 2020-10-09:
Here is what's at the end of memory:
```
00000000045af620 D _static_kernel_objects_end
00000000045b0000 R _binary_pagetables_bin_start
00000000045b0000 R z_x86_pagetables_start
00000000049b3000 R z_x86_kernel_ptables
00000000049b4000 R _binary_pagetables_bin_end
00000000049b4000 d wordlist
00000000049b4a00 d kernel_mutexes
00000000049b4a20 d asso_values.5025
00000000049b4b21 D _end
00000000049b4b21 D _image_ram_end
000000007fee6000 A __kernel_ram_size
0000000080000000 A __kernel_ram_end
```

the `kobject_data` section is an odd size, but I don't think it's incorrect:
```
 18 kobject_data  00000b21  00000000049b4000  00000000049b4000  00426000  2**5
                  CONTENTS, ALLOC, LOAD, DATA
```

andrewboie, 2020-10-09:
@jhedberg FYI, you might run into this at some point

andyross, 2020-10-12:
FWIW: the only reason this assert exists is because I was lazy in the C code and just wrote the section copies as a loop copying 64 bit values (and I thought, incorrectly, that ELF guaranteed 8 byte section sizes).  We could replace that with a cleaner memcpy() implementation, or frankly just a loop over bytes, and get rid of this.

andyross, 2020-10-13:
Can you see if the linked PR works for you?  My up_squared is still rigged for SOF work and will take a bit to untangle, and this is simple enough to have a high probability of working when coded blind.

nashif, 2020-10-13:
@aasthagr  @jenmwms  can you please verify this fix?

jenmwms, 2020-10-13:
@nashif Sure! I checked out that PR #29167 with the latest on main. Tested on HW using `west build -p -b up_squared tests/kernel/mem_protect/mem_protect`. I followed the Zephyr project Up Squared instructions, no modifications. SUCCESSFUL.
[test29167_memprotect_upsquared.txt](https://github.com/zephyrproject-rtos/zephyr/files/5373874/test29167_memprotect_upsquared.txt)

PR #29167 (2020-10-13): arch/x86: Make EFI copies bytewise
Originally the EFI boot code was written to assume that all sections
in the ELF file were 8-byte aligned and sized (because I thought this
was part of some platform spec somewhere).  This turned out to be
wrong in practice (at least for section sizes), so the requirement was
reduced to 4 bytes.  But now we have a section being generated
somewhere that turns out to violate even that.

There's no particular value in doing those copies in big chunks.
There's at best a mild performance benefit, but if we really cared
we'd be using a more complicated memcpy() impl [... cut]
commit 55a85771 (2020-10-13): arch/x86: Make EFI copies bytewise

Originally the EFI boot code was written to assume that all sections
in the ELF file were 8-byte aligned and sized (because I thought this
was part of some platform spec somewhere).  This turned out to be
wrong in practice (at least for section sizes), so the requirement was
reduced to 4 bytes.  But now we have a section being generated
somewhere that turns out to violate even that.

There's no particular value in doing those copies in big chunks.
There's at best a mild performance benefit, but if we really cared
we'd be using a more complicated memcpy() imp [... cut]
closed 2020-10-13 by andrewboie