zephyrproject-rtos/zephyr #29098: ATT timeout worker not canceled by disconnect, and may operate on disposed object

mtpr-ot, 2020-10-10:
If `att_timeout` is envoked after a disconnect, the "att" structure fetched from the work item has become invalid. This in turn causes an assertion in `bt_conn_index` via `gatt_ccc_conn_unqueue`.
The solution seems to be to cancel the delayed work in `att_reset` via `bt_att_disconnected`.

**To Reproduce**
Steps to reproduce the behavior:
1. Have an outstanding att request
2. Disconnect channel
3. Wait 30 sec
4. Assertion in `bt_conn_index` (conn pointer points to top of att_slab)

**Expected behavior**
Disconnection has been notified, do no further - no assert on timeout

**Impact**
This is seen multiple times on **v2.2** branch, in vendor configuration, and considered a show stopper.
Temporary solution in vendor 2.2.1 branch is to cancel work on disconnect. It does not appear (from code inspection) to be fixed on master.

mtpr-ot, 2020-10-11:
Updated text to reflect that it is more likely to be a "destroy" scenario, than a "disconnect" scenario. No hard evidence found yet to make this more clear, but I will update if we're able to obtain the information.

carlescufi, 2020-10-22:
@Vudentz could you please take a look at this one?

mtpr-ot, 2020-10-22:
@kapedev, please also review and update with your latest findings. I believe this issue may be v2.2-only.

carlescufi, 2020-11-05:
> @kapedev, please also review and update with your latest findings. I believe this issue may be v2.2-only.

@kapedev any updates on this? It would be very useful to know if this only affects 2.2 or is an issue in master.

klauspetersen, 2020-11-10:
An important observations was that when we hit the assert in _bt_conn_index_, the value of the handle _bt_conn_ *conn was the value of an _att_slab_ instead of a valid _bt_conn_ pointer. 
On 2.2, the _bt_conn_ handle is stored in the very top of the _att_slab_ that contains the handle. The _k_mem_slab_free_ will overwrite the first word of the _att_slab_ with a pointer to the next free slab, when freed. This leads me to believe that what we are seeing is a use-after-free type error.

From our trace, we know that the assert call stack ends with _gatt_ccc_conn_unqueue_() --> _bt_conn_index_(). I have gone trough the callee's of _gatt_ccc_conn_unqueue_ and my suspicion falls on the posibility of the  rtx_work delayed work which will execute the following
l2cap_rtx_timeout(struct k_work *work)
    bt_l2cap_chan_del(struct bt_l2cap_chan *chan)
        bt_att_disconnected(struct bt_l2cap_chan *chan)
            bt_gatt_disconnected(ch->chan.conn);
                gatt_ccc_conn_unqueue(conn);
                   bt_conn_index(bt_conn *conn)
                        Assert

If a channel has been destroyed (Free'd) without canceling the delayed rtx_work, then the above sequence would lead to a use-after-free.

The destroy function of a channel depends on how _l2cap_chan_add_ was called during the creation of that channel.
For fixed channels it is _bt_att_destroy_ 
For others it is _l2cap_chan_destroy_

Only _l2cap_chan_destroy_ calls cancel on rtx_work. So my suspicion is that rtx_work is run after a fixed channel has been destroyed. However, @Vudentz has informed me that this should only happen on dynamic channels.

carlescufi, 2020-11-12:
@Vudentz could you provide some feedback on @klauspetersen comment?

carlescufi, 2020-11-26:
@Vudentz ping

carlescufi, 2020-12-03:
@mtpr-ot and @klauspetersen have you done any further investigation on this?

carlescufi, 2021-01-07:
@Vudentz and @klauspetersen should we look at fixing this in time for 2.5?

carlescufi, 2021-01-14:
@joerchan please check if this is already fixed or will be fixed by an existing PR.

klauspetersen, 2021-01-14:
> @mtpr-ot and @klauspetersen have you done any further investigation on this?

I have not pursued this further. We hacked a small fix that made sure to cancel all delayed work, also any work remaining on rtx_work. We have not seen the issue since, but it was a very hard bug to reproduce in the first place.

joerchan, 2021-01-15:
@klauspetersen Would you mind sharing the "hack" you did that supposedly fixed the issue.

As far as I can see the delayed work is canceled in this case, but due to the way delayed work behaves a cancel is not guaranteed to succeed. It appears that ATT handles the case where the Timeout handler is invoked before the disconnected handler, by unassigned the L2CAP conn pointer, which prevents the disconnected handler from being invoked at all.
However if the disconnected handler is invoked first, and then the timeout handler the timeout handler will be run on an already detached ATT channel. In this case memory has been freed, and pointers assigned to zero, but not checked, this will lead to crash.

This is similar to the fix here:  https://github.com/zephyrproject-rtos/zephyr/pull/30655, the same fix should be applied to ATT timeout and ATT disconnected handlers.

As far as I can see the issue is the same on both master and 2.2.0 branch, but the rest of ATT is quite different so the crash would likely be different, and the fix is not easily backported to 2.2-branch.

PR #31357 (2021-01-25): Bluetooth: host: Handle ATT timeout on disconnected ATT channel 
Handle ATT timeout on disconnected ATT channel.
When the ATT channel is disconnected with a pending request the ATT
timeout is canceled and the response processed with an error code.
However canceling of delayed work is not guaranted to succeed, e.g:
"Work queue thread has removed the work item from the queue but has not
 called its handler"
This could lead to timeout handler being invoked after the disconnected
handler.

Fixes: #29098

Remove ATT modifying the L2CAP channel state by unassigning the
the connection pointer on time [... cut]
commit 28bddf93 (2021-01-25): Bluetooth: host: Handle ATT timeout on disconnected ATT channel

Handle ATT timeout on disconnected ATT channel.
When the ATT channel is disconnected with a pending request the ATT
timeout is canceled and the response processed with an error code.
However canceling of delayed work is not guaranted to succeed, e.g:
"Work queue thread has removed the work item from the queue but has not
 called its handler"
This could lead to timeout handler being invoked after the disconnected
handler.

Fixes: #29098
closed 2021-01-25 by nashif