zephyrproject-rtos/zephyr #29099: net: dns: dns-sd: support for dns service discovery

cfriedt, 2020-10-10:
**Is your enhancement proposal related to a problem? Please describe.**
Zephyr currently supports [mDNS](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/lib/dns/mdns_responder.c), which allows devices to be discovered on the network.

e.g. 
```bash
$ avahi-resolve -6 -n zephyr.local
zephyr.local	2001:db8::1
```

However, there is no way to discover which services are offered by a device yet (e.g. `avahi-browse` does not return anything for "zephyr").

If we were running an embedded HTTP server on a Zephyr device, we would expect something of the following form:

```bash
$ avahi-browse _http._tcp
+ lowpan0 IPv6 zephyr                                     Web Site             local
```

**Describe the solution you'd like**

There are a number of APIs in existence for DNS-SD. Most of them allow you to register an "empty" service descriptor (i.e. one without any tags).

E.g. one can [register a service](https://macchina.io/docs/00200-DNSSDTutorialAndUserGuide.html) with [macchina.io](https://macchina.io/docs/Poco.DNSSD.Service.html) like so

```cpp
Poco::DNSSD::DNSSDResponder dnssdResponder;
dnssdResponder.start();

Poco::DNSSD::Service service("_http._tcp", 8080);
Poco::DNSSD::ServiceHandle serviceHandle = dnssdResponder.registerService(service);
```

**Describe alternatives you've considered**

**Additional context**

jukkar, 2020-10-10:
This feature would be nice if you have time to implement it. :+1:

jukkar, 2020-10-10:
> * require dynamic memory allocation, or

Please no dynamic memory allocation. We can do it the same way as other similar network functionality is done. So let the user to decide how many services to support and allocate that at compile time. The API would then just return -ENOENT if there are no service "slots" available. We could provide a static service allocation macros and also an API to dynamically allocate a service. In both cases, the service would start to announce itself after the socket is bound by bind() call. We probably need to allow user to mark an already bound socket to be discoverable.

> * inherent permission / security issues (would we really want this to be a syscall?)

We need to support usermode too.

cfriedt, 2020-10-10:
> > * require dynamic memory allocation, or
> 
> Please no dynamic memory allocation. We can do it the same way as other similar network functionality is done.

I think I won't even use dynamic memory allocation for a PoC. I may have miscommunicated, but I was arguing **against** that.

It may be easy enough to use static / const defines declared to be part of a certain rom section that is iterable.

> We probably need to allow user to mark an already bound socket to be discoverable.

Technically speaking, the advertisement of a service is not really tied to its functionality, so no need to even check. Zephyr has no registry of running services / claimed ports. Probably out of scope for this item.

> > * inherent permission / security issues (would we really want this to be a syscall?)
> 
> We need to support usermode too.

Note, this is not a service discovery application, only the "service advertisement service", so it's likely that usermode is not necessary as long as services are registered at compile-time.

However, just in the process of analyzing what my is on my personal network with a Google Home Mini and a Chromecast, it is very evident that Google is adjusting mDNS and DNS-SD records dynamically (you and by extension, everyone, can see what is playing on your Chromecase with `avahi-browse -r _googlecast._tcp`).

cfriedt, 2020-10-12:
I went over my implementation about 100x and then I remembered this.

https://tools.ietf.org/html/rfc1123#page-73
```
         6.1.2.4  Compression: RFC-1035 Section 4.1.4

            Name servers MUST use compression in responses.

            DISCUSSION:
                 Compression is essential to avoid overflowing UDP
                 datagrams; see Section 6.1.3.2.
```

This makes things slightly more complicated. Still doable. Just a bit of an annoyance.

jukkar, 2020-10-13:
> This makes things slightly more complicated. Still doable. Just a bit of an annoyance.

Yes, DNS compression is quite annoying feature. We have some functions in `subsys/net/lib/dns/dns_pack.*` to handle it.

Gulruhbet, 2020-10-21:
can you write the code here

PR #29179 (2020-11-10): net: dns: dns-sd: support dns service discovery
This change adds support for DNS Service Discovery (DNS-SD) as described in RFC 6763. It also adds DNS-SD support to the `mdns_responder` service and sample application as well as a function call `net_context_port_in_use()`. 

Fixes #29099
Fixes #29429
Fixes #29649
commit 0ddce66d (2020-11-10): tests: net: dns: dns-sd: tests for dns service discovery

Tests for DNS-SD (RFC 6763)

Fixes #29099
commit e7e58439 (2020-11-10): net: dns: dns-sd: support dns service discovery

This change adds support for DNS Service Discovery (DNS-SD)
as described in RFC 6763.

Fixes #29099
closed 2020-11-10 by andrewboie