zephyrproject-rtos/zephyr #29107: Bluetooth: hci-usb uses non-standard interfaces

pabigot, 2020-10-11:
The Bluetooth specification (5.2v4B2.1.1) requires that a USB HCI primary controller provide the bulk and interrupt endpoints as interface 1, and a scalable isochronous endpoint at interface 2.  The latter supports the synchronous connection oriented capability required for Bluetooth audio.

Zephyr doesn't have the ability to specify which interface the BT HCI controller ends up at, and it doesn't provide an SCO endpoint at all.

Linux [btusb](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/bluetooth/btusb.c#n3838) verifies this requirement.  And it happens that if the only USB interface in the Zephyr application is the HCI one, `samples/bluetooth/hci_usb` works fine as a Linux bluetooth HCI device.

If some other devices are added, for example a CDC-ACM intended to serve as a shell, or a console that can be used to get log messages out, it does not.  This appears to be because Zephyr sorts the candidate interface descriptors by size, and the HCI one comes after the others, breaking the validation check in Linux.

While it's possible to [patch the btusb module](https://gist.github.com/pabigot/617f680a6ab89b3c7c643044ae9c758e) to bypass this check (an existing exception is in place for Apple Broadcom), the resulting interface is not usable:

```
raspi3bp[92]$ hciconfig
hci1:	Type: Primary  Bus: USB
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN INIT RUNNING 
	RX bytes:0 acl:0 sco:0 events:0 errors:0
	TX bytes:0 acl:0 sco:0 commands:1 errors:1
```

My theory is this is because Linux issues the probe while Zephyr is busy setting up the earlier interfaces.

There should be a way to ensure that the HCI-USB implementation conforms to the specification:

* The existing bulk HCI-USB interface should be installed as interface 1;
* A new an empty interface number 2 added that can be detected by btusb for the ISOC endpoint.

IMO that part's a bug.  If, an enhancement may be necessary to delay announcing existence of the interface to Linux until Zephyr's ready to respond to the on-probe commands.

May relate to the original goal of #4767

jfischer-phytec-iot, 2020-10-11:
>The Bluetooth specification (5.2v4B2.1.1) requires that a USB HCI primary controller provide the bulk and interrupt endpoints as interface 1, and a scalable isochronous endpoint at interface 2. The latter supports the synchronous connection oriented capability required for Bluetooth audio.

Actually, first interface and second interface, I did not found where bInterfaceNumbers are claimed to be specific value. This would also defeat the purpose of an interface descriptor.

>Zephyr doesn't have the ability to specify which interface the BT HCI controller ends up at, and it doesn't provide an SCO endpoint at all.

Zephyr does not need it, nobody needs it, that is what the descriptors are for. But within the local descriptor content of a class you can define the order of interfaces. 

>Linux btusb verifies this requirement. And it happens that if the only USB interface in the Zephyr application is the HCI one, samples/bluetooth/hci_usb works fine as a Linux bluetooth HCI device.

If btusb has problems finding the right interface then it is a problem in btusb not Zephyr. But if remember correctly we can change the order on the Zephyr side so that hci_usb always appears as first entity.

>the resulting interface is not usable:

>raspi3bp[92]$ hciconfig
hci1:	Type: Primary  Bus: USB
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN INIT RUNNING 
	RX bytes:0 acl:0 sco:0 events:0 errors:0
	TX bytes:0 acl:0 sco:0 commands:1 errors:1

>My theory is this is because Linux issues the probe while Zephyr is busy setting up the earlier interfaces.

My theory is this is because the function to generate a random address requires "User-space interface for hash algorithms"
from Linux kernel. This option is not enabled in the RPi kernel image (tested 4.19.85-v7+) and the BlueZ failes to bring up the device.

pabigot, 2020-10-11:
For most of that, all I can say is that the specification appears to be pretty clear.

> My theory is this is because the function to generate a random address requires "User-space interface for hash algorithms"
> from Linux kernel. This option is not enabled in the RPi kernel image (tested 4.19.85-v7+) and the BlueZ failes to bring up the device.

No, it's not that.  `CONFIG_CRYPTO_USER_API_HASH=m` is set in the 5.4.69 config I'm using under Yocto, and the driver works just fine if the first interface has `bInterfaceNumber=0` as required.

jfischer-phytec-iot, 2020-10-11:
> For most of that, all I can say is that the specification appears to be pretty clear.
> if the first interface has `bInterfaceNumber=0` as required.

Where did you get this condition that bInterfaceNumber must be 0, where is it in the specification?

You can force hci_usb class to be first element with:
```diff
diff --git a/subsys/usb/class/bluetooth.c b/subsys/usb/class/bluetooth.c
index 5511b24dc0..5982784efd 100644
--- a/subsys/usb/class/bluetooth.c
+++ b/subsys/usb/class/bluetooth.c
@@ -50,7 +50,7 @@ struct usb_bluetooth_config {
        struct usb_ep_descriptor if0_in_ep;
 } __packed;
 
-USBD_CLASS_DESCR_DEFINE(primary, 0)
+USBD_CLASS_DESCR_DEFINE(0primary, 0)
        struct usb_bluetooth_config bluetooth_cfg = {
        /* Interface descriptor 0 */
        .if0 = {
```

pabigot, 2020-10-11:
BLUETOOTH CORE SPECIFICATION Version 5.2 | Vol 4, Part B, section 2 "USB endpoint expectations".

> The Primary Controller configuration consists of two interfaces. The first
> interface has no alternate settings and contains the bulk and interrupt
> endpoints. The second interface provides scalable isochronous bandwidth.

I will grant you that first and second may not require interface numbers 0 and 1, and it's just chance that in the twelve years Linux has had HCI-USB driver support, with dozens of explicitly supported devices and uncounted standard ones, only one device hasn't used interface numbers 0 and 1.

In any case, if that isn't a hard requirement of the specification, I've already worked around that, but it'll still hit anybody who can't build their own custom driver in Linux.

> You can force hci_usb class to be first element with:

No, as you rediscovered, it isn't, because the entries are sorted by size first.

I do think that if I could make sure that the bluetooth stuff was up and running before Linux tries to communicate with the endpoints it might work.

jfischer-phytec-iot, 2020-10-11:
> > You can force hci_usb class to be first element with:
> 
> No, as you rediscovered, it isn't, because the entries are sorted by size first.

Other way around, you can try to force CDC_ACM to be behind hci_usb:
```diff
diff --git a/subsys/usb/class/cdc_acm.c b/subsys/usb/class/cdc_acm.c
index 1d1fb0b191..2bd6c61b05 100644
--- a/subsys/usb/class/cdc_acm.c
+++ b/subsys/usb/class/cdc_acm.c
@@ -1045,7 +1045,7 @@ static const struct uart_driver_api cdc_acm_driver_api = {
 
 #if (CONFIG_USB_COMPOSITE_DEVICE || CONFIG_CDC_ACM_IAD)
 #define DEFINE_CDC_ACM_DESCR(x, int_ep_addr, out_ep_addr, in_ep_addr)  \
-       USBD_CLASS_DESCR_DEFINE(primary, x)                             \
+       USBD_CLASS_DESCR_DEFINE(primary, 1.##x)                         \
        struct usb_cdc_acm_config cdc_acm_cfg_##x = {                   \
        .iad_cdc = INITIALIZER_IAD,                                     \
        .if0 = INITIALIZER_IF(0, 1, COMMUNICATION_DEVICE_CLASS,         \
@@ -1070,7 +1070,7 @@ static const struct uart_driver_api cdc_acm_driver_api = {
 }
 #else /* (CONFIG_USB_COMPOSITE_DEVICE || CONFIG_CDC_ACM_IAD) */
 #define DEFINE_CDC_ACM_DESCR(x, int_ep_addr, out_ep_addr, in_ep_addr)  \
-       USBD_CLASS_DESCR_DEFINE(primary, x)                             \
+       USBD_CLASS_DESCR_DEFINE(primary, 1.##x)                         \
        struct usb_cdc_acm_config cdc_acm_cfg_##x = {                   \
        .if0 = INITIALIZER_IF(0, 1, COMMUNICATION_DEVICE_CLASS,         \
                              ACM_SUBCLASS),                     
```

You can check the order of the entities in zephyr.map, for example for the hid-cdc sample:
```
*(SORT_BY_NAME(SORT_BY_ALIGNMENT(.usb.descriptor*)))                                         
 .usb.descriptor_primary.0                                                                    
                0x0000000020000488       0x1b libzephyr.a(usb_descriptor.c.obj)               
 .usb.descriptor_primary.1.0                                                                  
                0x00000000200004a3       0x19 subsys/usb/class/hid/libsubsys__usb__class__hid.a(core.c.obj)                                                                                 
 .usb.descriptor_primary.1.1                                                                  
                0x00000000200004bc       0x19 subsys/usb/class/hid/libsubsys__usb__class__hid.a(core.c.obj)                                                                                 
 .usb.descriptor_primary.1.1.0                                                                
                0x00000000200004d5       0x42 libzephyr.a(cdc_acm.c.obj)                      
 .usb.descriptor_primary.1.1.1                                                                
                0x0000000020000517       0x42 libzephyr.a(cdc_acm.c.obj)                      
 .usb.descriptor_primary.4                                                                    
                0x0000000020000559       0x70 libzephyr.a(usb_descriptor.c.obj)               
 .usb.descriptor_primary.5                                                                    
                0x00000000200005c9        0x2 libzephyr.a(usb_descriptor.c.obj)               
                0x00000000200005cb                __usb_descriptor_end = .      
```

carlescufi, 2021-02-04:
@holtmann could you please weigh in on this discussion?

pabigot, 2021-04-21:
It looks like Zephyr now supports ISO bluetooth operations, so maybe this is still worth doing?

jfischer-no, 2021-05-04:
> It looks like Zephyr now supports ISO bluetooth operations, so maybe this is still worth doing?

AFAIK BR/EDR USB audio interface cannot be used for ISO or not yet.

jfischer-no, 2021-05-04:
I leave https://github.com/zephyrproject-rtos/zephyr/issues/29107#issuecomment-706770001 as workaround and will not fix it for now.

carlescufi, 2021-05-19:
A more generic solution beyond workaround posted above will come in the future under a USB rework.

closed 2021-05-19 by carlescufi

pabigot, 2021-05-19:
My opinion for what it's worth:

While there's a proposed workaround that requires modifying the Zephyr OS code itself, the problem remains: it is not confirmed that Zephyr's behavior is consistent with the specification, and a dongle application combining a Zephyr HCI-USB class with other USB classes like a console or shell **will not work on Linux without a change to Zephyr's USB framework**.

That a solution "will come in the future" suggests it's still recognized as a bug (or an enhancement) that really does need to be addressed.  However, because this is closed, there's nothing to help ensure that future work will actually provide a solution.

Closing this as "Wont Fix" with a comment acknowledging intent to fix it later may reduce the bug census, but it doesn't make sense from a software quality process point of view.

konovagit, 2022-02-21:
what is the news on this issue ?

in the most recent release the workaround on subsys/usb/class/cdc_acm.c seem not work anymore.

I try to modify on line 1031
`USBD_CLASS_DESCR_DEFINE(primary, 1.##x)				\`

and check the zephyr.map
```
 *(SORT_BY_NAME(SORT_BY_ALIGNMENT(.usb.descriptor*)))
 .usb.descriptor_primary.0
                0x000000002000090c       0x1b zephyr/libzephyr.a(usb_descriptor.c.obj)
 .usb.descriptor_primary.1.0
                0x0000000020000927       0x1e zephyr/libzephyr.a(bluetooth.c.obj)
 .usb.descriptor_primary.1.1.0
                0x0000000020000945       0x42 zephyr/libzephyr.a(cdc_acm.c.obj)
 .usb.descriptor_primary.4
                0x0000000020000987       0x58 zephyr/libzephyr.a(usb_descriptor.c.obj)
 .usb.descriptor_primary.5
                0x00000000200009df        0x2 zephyr/libzephyr.a(usb_descriptor.c.obj)
                0x00000000200009e1                __usb_descriptor_end = .
```

some suggestions ?

konovagit, 2022-02-21:
Error on btusb:
```
[   38.262123] cdc_acm 1-1.4:1.1: ttyACM0: USB ACM device
[   38.262288] usbcore: registered new interface driver cdc_acm
[   38.262301] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   38.262406] btusb: probe of 1-1.4:1.0 failed with error -16
[   38.262503] usbcore: registered new interface driver btusb
```

jfischer-no, 2022-02-22:
The workaround to change what class interface appears at the end still works, but that will not help you with btusb, since it simple does not work with composite configuration.

```diff c
diff --git a/subsys/usb/class/cdc_acm.c b/subsys/usb/class/cdc_acm.c
index 9adc89a9c4b..df207d4a3a0 100644
--- a/subsys/usb/class/cdc_acm.c
+++ b/subsys/usb/class/cdc_acm.c
@@ -1010,7 +1010,7 @@ static const struct uart_driver_api cdc_acm_driver_api = {
        }
 
 #define CDC_ACM_CFG_AND_DATA_DEFINE(x)                                 \
-       USBD_CLASS_DESCR_DEFINE(primary, x)                             \
+       USBD_CLASS_DESCR_DEFINE(primary, 1.##x)                         \
        struct usb_cdc_acm_config cdc_acm_cfg_##x = {                   \
                INITIALIZER_IAD                                         \
                .if0 = INITIALIZER_IF(0, 1,                             \
```

konovagit, 2022-02-22:
So it's not possible to use HCI USB + Console or shell at the same time ? I don't understand for which problem the workaround works.

konovagit, 2022-02-23:
What can I patch on the zephyr side to avoid changing my btusb driver?

jfischer-no, 2022-02-23:
> So it's not possible to use HCI USB + Console or shell at the same time ? I don't understand for which problem the workaround works.

_The workaround to change what class interface appears at the end still works_

> What can I patch on the zephyr side to avoid changing my btusb driver?

Nothing, you cannot use hci_usb and CDC ACM in composite configuration with btusb.