zephyrproject-rtos/zephyr #29111: Atmel SAM V71 UART_0 fail

y2sh, 2020-10-12:
**Describe the bug**
Hi, I have an eval board of Atmel - SAMV71 Xplained. I succeed to download the sample code of 'Hello Word', it works great.
However, when I tried to change the console to uart0 - the code stuck at fatal.c , without recover.

**To Reproduce**
Add the following line in "prj.conf": 
CONFIG_UART_CONSOLE_ON_DEV_NAME="UART_0"

Add file "v71.overlay":
&uart0 {
     status = "okay";
	 current-speed = < 0x1c200 >;
};

Add the following line in "CMakeLists.txt":
set(DTC_OVERLAY_FILE v71.overlay)

The changes are taking effect on the build output - I can see it in "\build\zephyr\.config" and in "\build\zephyr\zephyr.dts"

**Expected behavior**
The reason I need the output in uart0 is because I have other custom board with v71 which it's output console in uart0. After lots of unsuccessful tries I go back to the eval board, and also in the eval board it failed - stuck in fatal.c

**Impact**
showstopper

**Environment**
 - OS: Windows
 - Version: BUILD_VERSION=zephyr-v2.4.0-468-g3caba2d52765

y2sh, 2020-10-12:
ok, I did some debug work in the meantime, it appear that in 'UART_0' - 'uart_console_dev' is null, in 'uart_console.c' , while when in the original settings it is not. what am I missing ?

y2sh, 2020-10-12:
I did some more debugging - it appear that only usart_1 is define in 'device' - I saw that in function 
'const struct device *z_impl_device_get_binding(const char *name)' in 'device.c'
 - so , how do I add uart_0 to devices ?

nandojve, 2020-10-12:
Hi @y2sh . I'm not complete sure if there is a bug. Let's check how do you should define at board first.

SAM's are using pinctrl to manage pins. In the case of UART0, pins are already defined here:
https://github.com/zephyrproject-rtos/zephyr/blob/c765b92944f2f01b611cb0696b4b9c073025588f/dts/arm/atmel/same70.dtsi#L143-L151

Then, at board level, you need define the proper UART port on the device tree. Let's check how we defined USART1 for SAM_V71:
https://github.com/zephyrproject-rtos/zephyr/blob/c765b92944f2f01b611cb0696b4b9c073025588f/boards/arm/sam_v71_xult/sam_v71_xult-common.dtsi#L21-L23

finally, you need enable the node and set required speed:
https://github.com/zephyrproject-rtos/zephyr/blob/c765b92944f2f01b611cb0696b4b9c073025588f/boards/arm/sam_v71_xult/sam_v71_xult-common.dtsi#L171-L174

Could you try define above steps and build hello world?

y2sh, 2020-10-13:
Hi, thanks for your reply. I did all the changes you mentioned- not in the source files, but in the overlay as recommended in zephyr docs. as I wrote - I can see that the overlay take effect in the output 'zephyr.dts' file in the build directory. I only missed the 'chosen' field, thanks for notify that. anyhow - the code still could not bind uart0, it is not apear in the device struct array - you can see in the screenshot (debug with vscode). what else could it be?
[image]
[image]
[image]

y2sh, 2020-10-13:
also this screenshot:
[image]

y2sh, 2020-10-13:
YES !! I solved it !
I looked for hints in the .config output file, and find out that I had to set also 'CONFIG_UART_SAM=y' . 
thanks anyway.

closed 2020-10-13 by y2sh

nandojve, 2020-10-13:
Ahh,,,sure forgot to mention :D , we use Kconfig to define the rule that builds the files:

https://github.com/zephyrproject-rtos/zephyr/blob/8d563d65074c26e6a19cace753fe9558fd28f323/drivers/serial/CMakeLists.txt#L24-L25

So, yes, you need enable the desired driver there. It can be by a conf or a defconfig file, since you are using overlays. Try look at
https://github.com/zephyrproject-rtos/zephyr/tree/master/boards/shields/esp_8266 to get inspirations. You can search on samples and look at CMakes to understand better the mechanism.

Now it is working, I would recommend you "copy" the SAM_V71_XULT board and rename to your own board. Then add your changes there. It will be easier add things on one place instead add overlays/conf on every sample/app that you will touch. Anyway, it is just a suggestion.