zephyrproject-rtos/zephyr #29147: Store logger filtering data in persistent memory

nordic-krch, 2020-10-13:
**Is your feature request related to a problem? Please describe.**
Logger supports runtime filtering of log messages. Filtering is independent for each backend so you can have more logs on uart backend but less on file system crash log backend. Filtering can be configured by shell commands but after reboot current configuration is lost and same set of commands needs to be applied to get back to previous filtering.

**Describe the solution you'd like**
Provide an API (`log_config_store()`/`log_config_restore()`) and shell commands for storing and restoring filtering information. Filters are stored in one memory section so data is there.

nordic-krch, 2020-10-13:
@nvlsianpu as zephyr supports multiple persistent storage options how would you see it being solved in a generic way? Kconfig to decide which location to use?

nvlsianpu, 2020-10-15:
settings subsystem is something which can store such a configuration.
I would say that if settings is in usage log_config_store()/log_config_restore(),
also pure nvs() implementation is possible if settings solution is to much (it even can be used while settings uses nvs)
alternatively user might provide simpler implementation - especiali if device has backup register or an high endurance memory.

drandreas, 2020-10-27:
I've a question that goes into a similar direction. Would it be possible to have the default log level per log sources independent of the compiled log level? Currently `log_core.c` copies the compiled log level into the runtime filter:

```
if (IS_ENABLED(CONFIG_LOG_RUNTIME_FILTERING)) {
	for (int i = 0; i < log_sources_count(); i++) {
		uint32_t *filters = log_dynamic_filters_get(i);
		uint8_t level = log_compiled_level_get(i);

		LOG_FILTER_SLOT_SET(filters,
			    LOG_FILTER_AGGR_SLOT_IDX,
			    level);
	}
}
```

I see use cases for having more logging available but muted during normal operation.

One issue I see with `log_config_store()/log_config_restore()` is the compile time coupling of sources to indices. So if a module is altered the index of a source might be shifted by an offset.