zephyrproject-rtos/zephyr #29148: MPU: twr_ke18f: many kernel application fails when allocate dynamic MPU region

hakehuang, 2020-10-13:
**Describe the bug**
E: Failed to allocate new MPU region 8

**To Reproduce**
Steps to reproduce the behavior:
cd tests/arch/arm/arm_ramfunc
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Expected behavior**
test pass

**Impact**
MPU hardware protect

impacted application: (which is OK in v2.4.0RC3)

1. arch.arm.swap.common.fpu_sharing
2. arch.arm.ramfunc
3. arch.arm.ramfunc.arch.arm.ramfunc
4. arch.arm.swap
5. arch.arm.swap.common
6. arch.arm.swap.common.arch.arm.swap.common
7. arch.arm.swap.common.fpu_sharing.arch.arm.swap.common.fpu_sharing
8. arch.arm.swap.common.no_optimizations
9. arch.arm.swap.common.no_optimizations.arch.arm.swap.common.no_optimizations
10. arch.arm.swap.common.fpu_sharing.no_optimizations
11. kernel.device
12. kernel
13. kernel.device.kernel.device
14. kernel.device.pm
15. kernel.device.pm.kernel.device.pm
16. kernel.pipe
17. kernel.queue
18. kernel.timer
19. kernel.memory_protection.userspace
20. kernel.memory_protection.userspace.kernel.memory_protection.userspace
21. kernel.futex
22. kernel.futex.kernel.futex
23. kernel.message_queue
24. kernel.message_queue.kernel.message_queue
25. kernel.mutex

**Logs and console output**

```
E: Failed to allocate new MPU region 8
* Booting Zephyr OS version 2.4.99  ***

Running test suite arm_ramfunc

===================================================================

START - test_arm_ramfunc

E: Failed to allocate new MPU region 8

ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/mpu/nxp_mpu.c:588

	Configuring 2 dynamic MPU regions failed

E: r0/a1:  0x00000004  r1/a2:  0x0000024c  r2/a3:  0x00800000

E: r3/a4:  0x0000000e r12/ip:  0x00000000 r14/lr:  0x0000252f

E:  xpsr:  0x6100000e

E: Faulting instruction address (r15/pc): 0x0000b4d6

E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0

E: Fault during interrupt handling

E: Current thread: 0x20000060 (unknown)

E: Halting system
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux,)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: 0d04754

hakehuang, 2020-10-13:
@MaureenHelm

henrikbrixandersen, 2021-01-18:
@hakehuang Is this still an issue after #31148 was merged?

hakehuang, 2021-01-19:
@henrikbrixandersen  it is ok, thanks for the fixes

closed 2021-01-19 by hakehuang