zephyrproject-rtos/zephyr #29155: CAN BUS support on Atmel V71

y2sh, 2020-10-13:
Hi, is there support for canbus in Atmel SAM V71 board ?

nandojve, 2020-10-13:
I think you can get a preview at #22832, maybe you need rebase. If you have conditions to test @alexanderwachter PR's it will help to upstream. I don't know if there are anyone working on CAN-BUS only.

CC: @alexanderwachter

martinjaeger, 2020-10-13:
Unfortunately, #22832 needs a bit more than a rebase, as there were several changes in Zephyr in between. We are working on PR #28345 at the moment. Once that one is merged, the actual driver integration for CAN-FD can continue. AFAIK the Atmel was the one that didn't work yet, so I guess that some development + testing support would be highly appreciated.

y2sh, 2020-10-14:
I am busy with other issues, so no development from my side for now. However, I can help with testing canbus relating code on my board.

closed 2020-12-28 by github-actions[bot]