zephyrproject-rtos/zephyr #29164: net: accept() doesn't return an immediately usable descriptor

ohitz, 2020-10-13:
**Describe the bug**

When a new connection on a socket is accepted with `accept()` and the returned file descriptor is immediately used to send data using `send()`, that call fails. Only if a small `k_sleep(K_MSEC(1));` is inserted after the `accept()` call, `send()` succeeds.

The file descriptor returned by `accept()` cannot be immediately used.

**To Reproduce**

1. Apply this patch to the `echo_server`: [echo_server.txt](https://github.com/zephyrproject-rtos/zephyr/files/5372238/echo_server.txt)
2. Compile and flash
3. Connect to the echo server with a telnet client: `telnet 192.0.2.1 4242`. You should see an error logged to the console but no welcome message.
4. Insert a `k_sleep(K_MSEC(1));` between `accept()` and `send()` and retest. You should now receive the `Hello client!` welcome message.

**Expected behavior**

A client connecting to the echo server should see a `Hello client!` message without any additional sleep.

**Environment:**

 - Zephyr fd059fe1f228cf45b3ddea205fa12e94a3dd252c
 - MIMXRT1060-EVK eval board

jukkar, 2020-10-13:
The patch echo_server.txt looks like this
```
diff --git a/samples/net/sockets/echo_server/src/tcp.c b/samples/net/sockets/echo_server/src/tcp.c
index 0ff35a1852..6cef2dae9f 100644
--- a/samples/net/sockets/echo_server/src/tcp.c
+++ b/samples/net/sockets/echo_server/src/tcp.c
@@ -234,6 +234,12 @@ static int process_tcp(struct data *data)
 
 	LOG_INF("TCP (%s): Accepted connection", data->proto);
 
+#define HELLO "Hello client!\r\n"
+	int sent = sendall(client, HELLO, sizeof(HELLO));
+	if (sent < 0) {
+		LOG_ERR("TCP(%s): initial send error %d", data->proto, sent);
+	}
+
 #if defined(CONFIG_NET_IPV6)
 	if (client_addr.sin_family == AF_INET6) {
 		tcp6_handler_in_use[slot] = true;
```

jukkar, 2020-10-13:
I have a fix proposal, but was not able to test it right now (can tomorrow). @ohitz if possible, please try the PR and report back if it works for you.

ohitz, 2020-10-14:
Thanks @jukkar I have tried it but unfortunately your PR doesn't solve the issue for me.

I have attached the debugger and found that internally a call to `net_context_send()` (in `/subsys/net/lib/sockets/sockets.c:611`) returns `-ENOTCONN`. Maybe that helps.

jukkar, 2020-10-14:
> Maybe that helps.

Thanks, this info helped a lot.

PR #29169 (2020-10-14): net: tcp2: Socket was accepted too early
The TCP2 was calling accept callback before actually finalizing
the connection attempt.

Fixes #29164
commit d3283231 (2020-10-14): net: tcp2: Socket was accepted too early

The TCP2 was calling accept callback before actually finalizing
the connection attempt.

Fixes #29164
closed 2020-10-14 by jukkar