zephyrproject-rtos/zephyr #29173: uart_nrfx_uart fails uart_async_api_test

pabigot, 2020-10-13:
The remaining failures are shown here:
<details><summary>Test failures for ready events</summary>

```
===================================================================
START - test_chained_write

    Assertion failed at ../src/test_uart_async.c:492: test_chained_write: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_chained_write
===================================================================
START - test_long_buffers_setup
 PASS - test_long_buffers_setup
===================================================================
START - test_long_buffers

    Assertion failed at ../src/test_uart_async.c:564: test_long_buffers: (k_sem_take(&rx_rdy, K_MSEC(200)) not equal to 0)
RX_RDY timeout
 FAIL - test_long_buffers
===================================================================
START - test_write_abort_setup
 PASS - test_write_abort_setup
===================================================================
START - test_write_abort

    Assertion failed at ../src/test_uart_async.c:350: test_write_abort: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_write_abort
===================================================================
START - test_forever_timeout_setup
 PASS - test_forever_timeout_setup
===================================================================
START - test_forever_timeout

    Assertion failed at ../src/test_uart_async.c:426: test_forever_timeout: (k_sem_take(&rx_rdy, K_MSEC(100)) not equal to 0)
RX_RDY timeout
 FAIL - test_forever_timeout
===================================================================
Test suite uart_async_test failed.
===================================================================
PROJECT EXECUTION FAILED
```
</details>

Serendipitously confirming my belief that the use of `k_delayed_work` there was wrong, these are fixed by #28891 which changes the driver to use to the new delayable work API.

_Originally posted by @pabigot in https://github.com/zephyrproject-rtos/zephyr/issue_comments/706385538_

PerMac, 2020-10-19:
Other tests from ./tests/drivers/uart/ also fail (i.e: all tests from uart_mix_fifo_poll)

anangl, 2020-10-20:
@pabigot I cannot reproduce those failures, even if I go back to the commit https://github.com/zephyrproject-rtos/zephyr/commit/4a0a196f574e6f5f0089b510c36ade4bf083eb29 that was merged in #29093 that closed #29092. All test cases pass, unless I remove the connection between P1.01 and P1.02. But then obviously even more test cases fail. Am I missing something?

pabigot, 2020-10-20:
It could be dependent on the relative errors in HFCLK and LFCLK.

pabigot, 2021-01-26:
Looks like the failures noted above are a result of not shorting P1.01 with P1.02, and I can't reproduce any problems with information available, so consider this fixed by subsequent changes to the test or infrastructure.

closed 2021-01-26 by pabigot