zephyrproject-rtos/zephyr #29195: west fails with custom manifest

endian-benjamin, 2020-10-14:
**Describe the bug**
West fails on newer upstream using a custom west.yml, which built fine on v2.3.0. 

It's hard to give exact reproduction recipe because I can't share the manifest currently. But the problem is essentially in `scripts/zephyr_module.py`:

```
    if args.modules is None:
        # West is imported here, as it is optional
        # (and thus maybe not installed)
        # if user is providing a specific modules list.
        from west.manifest import Manifest
        from west.util import WestNotFound
        try:
            manifest = Manifest.from_file()
            projects = [p.posixpath for p in manifest.get_projects([])]
            # ^--- this line
```

so `get_projects([])` manages to fetch the correct projects, but for whatever reason an additional `None` is prepended here, which it wasn't before. The fix was easy: 

```
            projects = [p.posixpath for p in manifest.get_projects([])
                        if p.posixpath]
```

but I don't know if it's strictly "correct" as far as fixes go.

endian-benjamin, 2020-10-14:
https://github.com/zephyrproject-rtos/zephyr/pull/29196

mbolivar-nordic, 2020-10-14:
Does this reproduce with west master?

mbolivar-nordic, 2020-10-15:
This was fixed in west 0.8. See https://github.com/zephyrproject-rtos/zephyr/pull/29196#issuecomment-709552722 for details.

closed 2020-10-15 by mbolivar-nordic