zephyrproject-rtos/zephyr #29199: github integration: ensure maintainers are added to PRs that affect them

pabigot, 2020-10-14:
I'm maintainer of I2C, but there are things getting labeled as I2C that I'm not being notified of.

Perhaps maintainers aren't supposed to be notified of anything that affects their area, but I would certainly like to know about changes to what I'm nominally responsible for.

pabigot, 2020-10-14:
Or that there's a disconnect between being a maintainer and being a code-owner.  There are 11 `CODEOWNERS` entries that are covered by the I2C `MAINTAINERS` patterns (and one that isn't, but should be).

I don't think I should have to add myself to every `CODEOWNERS` line involving I2C, and monitor that file for cases where it gets out of sync again.

carlescufi, 2020-10-15:
Unfortunately this is a consequence of using multiple mechanisms for maintainership tracking. We can switch from `CODEOWNERS`, which is automatically handled by GitHub, to a custom workflow that assigns maintainers based on the `MAINTAINERS` file, but AFAIK this is not being worked on yet. @nashif or @ioannisg do you know if we have plans for such a workflow?

stephanosio, 2022-05-23:
Fixed by https://github.com/zephyrproject-rtos/zephyr/issues/38725:

Maintainers are now added as an assignee on the relevant PRs.

closed 2022-05-23 by stephanosio

pabigot, 2022-05-23:
> Fixed by #38725:
> 
> Maintainers are now added as an assignee on the relevant PRs.

How is it appropriate to close this issue, given that #38725 is open and on hold, and #45865 identifies more problems with the solution that is implemented?

Don't feel obliged to answer on my behalf.  🤦

stephanosio, 2022-05-23:
#38725 has already been implemented and is currently functional as noted above.
#45865 has nothing to do with this.