zephyrproject-rtos/zephyr #29223: BLE one central connect multiple peripherals

mfinmuch, 2020-10-15:
Hello,
I am using zephyr 2.4.99 to develop a BLE program
Currently, I used central_hr and peripheral in the sample to load separately on two nrf52840s
Now the two boards can communicate with each other, can read and write
But now I want a central_hr to connect to more than two peripherals, but when I loaded the peripheral to the third board, my central_hr could not be connected to the second peripheral
In addition, I also added CONFIG_BT_MAX_CONN=3 in prj.conf, but it still doesn’t work.
I started_sacn() again after finding the UUID
However, after scanning, it will be found that the handles of the two peripherals are the same, and the central request will cause conflicts when reading, so how do you distinguish when the central wants to read data from one of them?
Or is there a sample program for one central to multiple peripherals?

Thanks,
Poyi Lin

joerchan, 2020-11-04:
Which handle is the same? connection handle / ATT handle?
What exactly does not work.

Need more information in order to answer this question.

carlescufi, 2020-11-26:
@mfinmuch please provide the info requested

mfinmuch, 2020-11-26:
> Which handle is the same? connection handle / ATT handle?
> What exactly does not work.
> 
> Need more information in order to answer this question.

Hello
I finally found that my ATT handle will be the same
But my connection handle will be determined according to the connection order
However, how to select a specific peripheral to use, whether to use it according to the handle of the link, is still not very clear.

joerchan, 2020-11-30:
@mfinmuch You get the connection object to the connections you create. You will have to manage these, as these are the ones used to select which connection you want to perform the read operation on.
As you can see the first argument to bt_gatt_read is this connection object, `bt_conn *conn`.
The central_hr sample is written to only support one connection at a time, and therefore it only keeps track of one connection object, `default_conn`.

mfinmuch, 2020-12-02:
@joerchan 
I have seen other people implement one central to mutiple peripheral in other questions
I referred from here [https://github.com/zephyrproject-rtos/zephyr/issues/21552](url)
I use this to modify, central will go to scan peripherals, after subscribing, in the notify function will receive the data sent by the peripheral, received and immediately use `bt_gatt_write `to send back to the peripheral directly
When I have more than two peripherals links, one of the peripherals is disconnected, it will print [UNSUBSCRIBED] and jump to the disconnect callback function, then clear the disconnected link, then do `attempt_scan()`
However, the connected peripheral cannot send the data to the central at this time, and the central will be stuck in the `attempt_scan()` of disconnect, and the transmission can only continue after the central finds the disconnected peripheral again

How to prevent central from getting stuck in diconnect's `attempt_scan()` so that the connected peripheral can still send information to central?

The disconnected callback function as follow:
```
static void disconnected(struct bt_conn *conn, u8_t reason)
{
	char addr[BT_ADDR_LE_STR_LEN];

	bt_addr_le_to_str(bt_conn_get_dst(conn), addr, sizeof(addr));

    u8_t conn_index = bt_conn_index(conn);
    if (conn_index == connecting_index) {
        connecting_index = -1;
        is_connecting = false;
    }
    printk("Disconnected: %s, index: %u, ref: %u, (reason 0x%02x)\n", addr, conn_index, conn->ref, reason);

    if (conn_index < 32) {
        if (connections[conn_index] == conn) {
            bt_conn_unref(connections[conn_index]);
            connections[conn_index] = NULL;
            conn_num--;
        }
    }

    attempt_scan();
}
```
 Thanks in advance,

joerchan, 2020-12-08:
@mfinmuch I'm not sure why it would be stuck in `attempt_scan`.
Are you able to share the code and which commit of zephyr you are on so that I can reproduce this issue locally to figure this one out?

mfinmuch, 2020-12-09:
> @mfinmuch I'm not sure why it would be stuck in `attempt_scan`.
> Are you able to share the code and which commit of zephyr you are on so that I can reproduce this issue locally to figure this one out?

Finally, I found out that my call_back notify_func will set _params->value_handle=0U_ when unsubscribe occurs.
I guess this is because the notifry handle of the connected peripheral was also cleared.
This part feels that zephyr is not well managed in multilink

joerchan, 2020-12-10:
Are you using the same subscribe parameters for all connections, that is not supported.
You will have to have separate subscribe parameters per subscription.

mfinmuch, 2020-12-10:
> Are you using the same subscribe parameters for all connections? That is not supported.
> You will have to have separate subscribe parameters per subscription.
So how can I avoid the disconnected peripheral from affecting the connected peripheral without canceling `params-value_handle=0`?
Then why cancel the` params->value_handle=0` so that the peripheral in the connection will not be stuck?

Thanks,
Poyi

joerchan, 2020-12-10:
When you subscribe you use a pointer to subscribe parameteres, you have to provide different pointers.

For example like this:

```
static struct bt_gatt_subscribe_params subscribe_params[CONFIG_BT_MAX_CONN];
...

struct bt_gatt_subscribe_params *params = &subscribe_params[bt_conn_index(conn)];

params->notify = notify_func;
params->value = BT_GATT_CCC_NOTIFY;
params->ccc_handle = attr->handle;

err = bt_gatt_subscribe(conn, params);
```

mfinmuch, 2020-12-11:
> When you subscribe you use a pointer to subscribe parameteres, you have to provide different pointers.
>
> For example like this:
>
> ```
> static struct bt_gatt_subscribe_params [...]

Thank you,
The method you provide can be used, which is much more reasonable than I canceled `params-value_handle=0`

Poyi

joerchan, 2020-12-11:
Great, I'll close this issue now then.

closed 2020-12-11 by joerchan