zephyrproject-rtos/zephyr #29235: Endless build loop after adding pinctrl dtsi

Delsian, 2020-10-15:
**Describe the bug**

After adding include with pinctrl definitions, build continuously restarts:
```
[0/1] "C:\Program Files\CMake\bin\cmake.exe" --regenerate-during-build -SC:\Eug\zephyrproject\locinox\relayboxz -BC:\Eug\zephyrproject\locinox\relayboxz\build
-- Application: C:/Eug/zephyrproject/locinox/relayboxz
-- Zephyr version: 2.4.99 (C:/Eug/zephyrproject/zephyr)
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: relaybox
-- Cache files will be written to: C:/Eug/zephyrproject/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/Program Files (x86)/GNUARM)
-- Found BOARD.dts: C:/Eug/zephyrproject/locinox/commonz/boards/arm/relaybox/relaybox.dts
-- Generated zephyr.dts: C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Eug/zephyrproject/locinox/relayboxz/Kconfig
Loaded configuration 'C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/.config'
No change to configuration in 'C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/.config'
No change to Kconfig header in 'C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Eug/zephyrproject/locinox/relayboxz/build
[0/1] "C:\Program Files\CMake\bin\cmake.exe" --regenerate-during-build -SC:\Eug\zephyrproject\locinox\relayboxz -BC:\Eug\zephyrproject\locinox\relayboxz\build
-- Application: C:/Eug/zephyrproject/locinox/relayboxz
-- Zephyr version: 2.4.99 (C:/Eug/zephyrproject/zephyr)
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: relaybox
-- Cache files will be written to: C:/Eug/zephyrproject/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/Program Files (x86)/GNUARM)
-- Found BOARD.dts: C:/Eug/zephyrproject/locinox/commonz/boards/arm/relaybox/relaybox.dts
-- Generated zephyr.dts: C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Eug/zephyrproject/locinox/relayboxz/Kconfig
Loaded configuration 'C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/.config'
No change to configuration in 'C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/.config'
No change to Kconfig header in 'C:/Eug/zephyrproject/locinox/relayboxz/build/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Eug/zephyrproject/locinox/relayboxz/build
[0/1] "C:\Program Files\CMake\bin\cmake.exe" --regenerate-during-build -SC:\Eug\zephyrproject\locinox\relayboxz -BC:\Eug\zephyrproject\locinox\relayboxz\build
-- Application: C:/Eug/zephyrproject/locinox/relayboxz
...

**To Reproduce**

Adding  line

    #include <st/f4/stm32f412r(e-g)tx-pinctrl.dtsi>

into custom board DTS file
Removing this line returns build back to normal behavior    

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Windows10
 - Toolchain  Zephyr SDK, GCC
 - Commit SHA 2837ac4409a1d188eb9d489903c9d41e9bf94494

Delsian, 2020-10-15:
The same behavior with sample files:
```
-- west build: generating a build system
-- Application: C:/Eug/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.99 (C:/Eug/zephyrproject/zephyr)
-- Found Python3: C:/Python/python.exe (found suitable exact version "3.8.2") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nucleo_f412zg
-- Cache files will be written to: C:/Eug/zephyrproject/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/Program Files (x86)/GNUARM)
-- Found BOARD.dts: C:/Eug/zephyrproject/zephyr/boards/arm/nucleo_f412zg/nucleo_f412zg.dts
-- Generated zephyr.dts: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Eug/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Eug/zephyrproject/zephyr/boards/arm/nucleo_f412zg/nucleo_f412zg_defconfig'
Merged configuration 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
Kconfig header saved to 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNUARM/bin/arm-none-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build
-- west build: building application
[0/1] Re-running CMake...
-- Application: C:/Eug/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.99 (C:/Eug/zephyrproject/zephyr)
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nucleo_f412zg
-- Cache files will be written to: C:/Eug/zephyrproject/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/Program Files (x86)/GNUARM)
-- Found BOARD.dts: C:/Eug/zephyrproject/zephyr/boards/arm/nucleo_f412zg/nucleo_f412zg.dts
-- Generated zephyr.dts: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Eug/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to configuration in 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
No change to Kconfig header in 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build
[0/1] Re-running CMake...
-- Application: C:/Eug/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.99 (C:/Eug/zephyrproject/zephyr)
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nucleo_f412zg
-- Cache files will be written to: C:/Eug/zephyrproject/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/Program Files (x86)/GNUARM)
-- Found BOARD.dts: C:/Eug/zephyrproject/zephyr/boards/arm/nucleo_f412zg/nucleo_f412zg.dts
-- Generated zephyr.dts: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Eug/zephyrproject/zephyr/Kconfig
```

After switching to 2.4 release (7a3b253ced7333f5c0269387a7f3ed1dee69739d) everything works fine.
```
-- west build: generating a build system
-- Application: C:/Eug/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.0 (C:/Eug/zephyrproject/zephyr)
-- Found Python3: C:/Python/python.exe (found suitable exact version "3.8.2") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: nucleo_f412zg
-- Cache files will be written to: C:/Eug/zephyrproject/zephyr/.cache
-- Found dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (C:/Program Files (x86)/GNUARM)
-- Found BOARD.dts: C:/Eug/zephyrproject/zephyr/boards/arm/nucleo_f412zg/nucleo_f412zg.dts
-- Generated zephyr.dts: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/devicetree_unfixed.h
Parsing C:/Eug/zephyrproject/zephyr/Kconfig
Loaded configuration 'C:/Eug/zephyrproject/zephyr/boards/arm/nucleo_f412zg/nucleo_f412zg_defconfig'
Merged configuration 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/.config'
Kconfig header saved to 'C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build/zephyr/include/generated/autoconf.h'
[... cut ...]
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files (x86)/GNUARM/bin/arm-none-eabi-gcc.exe
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Eug/zephyrproject/zephyr/samples/basic/blinky/build
-- west build: building application
[1/129] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[2/129] Generating include/generated/driver-validation.h
[3/129] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
...
```

Delsian, 2020-10-15:
The same with samples:
Commit db89bb7ef8649747ec73fddd1525a430533c5483 builds successfully;
Next commit dda6396c6503390f21c4aa623a81d8ab7210f7fd (adding include ) goes to endless loop.

tejlmand, 2020-10-15:
Sounds related to this:
https://github.com/zephyrproject-rtos/zephyr/issues/26112

I did a check on the generated dependency file that will be used by Ninja, and on Linux it contains:
```
/projects/github/ncs/modules/hal/stm32/dts/st/f4/stm32f412z(e-g)tx-pinctrl.dtsi \
```

but in windows I get:
```
C:\projects\github\ncs\modules\hal\stm32/dts/st/f4/stm32f412z(e-g)tx-pinctrl.dtsi \
```

Looking into why it uses `\` on windows for the module path, as that is a known source of error.

tejlmand, 2020-10-15:
Fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/29243

closed 2020-10-16 by nashif

Delsian, 2020-10-16:
@tejlmand Thank you!

PR #29243 (2020-10-16): scripts: ensuring posix path on module defined roots.
Fixes: #29235

This commit ensures that roots are being converted to posix paths.
This fixes the issue where windows style path (containing `\`) would
result in DTS dependency file to contain mixed style path separator and
thus causing Ninja to re-invoke CMake in an endless loop.
commit fef78795 (2020-10-16): scripts: ensuring posix path on module defined roots.

Fixes: #29235

This commit ensures that roots are being converted to posix paths.
This fixes the issue where windows style path (containing `\`) would
result in DTS dependency file to contain mixed style path separator and
thus causing Ninja to re-invoke CMake in an endless loop.