zephyrproject-rtos/zephyr #29244: k_thread_resume() can cause k_sem_take with K_FOREVER to return -EAGAIN

laurenmurphyx64, 2020-10-15:
**Describe the bug**
When one thread is blocks performing `k_sem_take` with `K_FOREVER` on a semaphore whose count is set to 0 and another thread calls `k_thread_resume` on it, the thread will resume and `k_sem_take` call will return `EAGAIN`, an error code that is normallly reserved for the waiting period timing out.

**To Reproduce**
Steps to reproduce the behavior:
1. `cd samples/hello_world`
2. Replace `samples/hello_world/src/main.c` with the following:
```
#include <zephyr.h>
#include <sys/printk.h>

K_SEM_DEFINE(BLOCKSEMA, 0, 1);

void block_thread(void)
{
	printk("Pre sema take\n");
	
	int code = k_sem_take(&BLOCKSEMA, K_FOREVER);
	
	printk("Post sema take %i\n", code);
}

K_THREAD_DEFINE(block_thread_id, 512, (k_thread_entry_t)block_thread, NULL, NULL,
		NULL, 1, 0, 0);

void main(void)
{ 
	printk("Main\n");

	k_thread_resume(block_thread_id);
}
```
3. `west build -b qemu_x86 -t run .`
4. See error

**Expected behavior**
Either `k_sem_take` should continue blocking since `k_sem_give` wasn't called, or it should produce a different error code per [the documentation](https://docs.zephyrproject.org/latest/reference/kernel/synchronization/semaphores.html#c.k_sem_take)?

**Impact**
Unexpected behavior, possibly confusing to users. 

**Logs and console output**
Output:
```
$ west build -b qemu_x86 -t run .
...
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200625_115407-9426dddc0a1f-zephyr
)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.4.0-634-g53cc090c39a5  ***
Main
Pre sema take
Post sema take -11
FAILED: zephyr/CMakeFiles/run 
cd /home/herbert/zephyr-external/zephyr/samples/hello_world/build && /home/herbert/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 9 -cpu qemu32,+nx,+pae -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -net none -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -icount shift=5,align=off,sleep=off -rtc clock=vm -kernel /home/herbert/zephyr-external/zephyr/samples/hello_world/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/herbert/.local/bin/cmake --build /home/herbert/zephyr-external/zephyr/samples/hello_world/build --target run
```

**Environment:**
 - OS: Ubuntu 18.04
 - Board: qemu_x86
 - Toolchain: Zephyr SDK 0.11.4
 - Commit 53cc090

nashif, 2020-10-16:
now that i read through this I think it is still a valid bug.

nashif, 2020-10-16:
possible fix:

```diff
diff --git a/kernel/include/ksched.h b/kernel/include/ksched.h
index e7dc034bd7..c81eac4064 100644
--- a/kernel/include/ksched.h
+++ b/kernel/include/ksched.h
@@ -105,6 +105,11 @@ static inline bool z_is_idle_thread_object(struct k_thread *thread)
 #endif /* CONFIG_MULTITHREADING */
 }

+static inline bool z_is_thread_suspended(struct k_thread *thread)
+{
+       return (thread->base.thread_state & _THREAD_SUSPENDED) != 0U;
+}
+
 static inline bool z_is_thread_pending(struct k_thread *thread)
 {
        return (thread->base.thread_state & _THREAD_PENDING) != 0U;
diff --git a/kernel/sched.c b/kernel/sched.c
index a5932070d2..09d14797a1 100644
--- a/kernel/sched.c
+++ b/kernel/sched.c
@@ -481,7 +481,13 @@ static inline void z_vrfy_k_thread_suspend(struct k_thread *thread)

 void z_impl_k_thread_resume(struct k_thread *thread)
 {
-       k_spinlock_key_t key = k_spin_lock(&sched_spinlock);
+       k_spinlock_key_t key;
+
+       if (!z_is_thread_suspended(thread)) {
+               return;
+       }
+
+       key = k_spin_lock(&sched_spinlock);

        z_mark_thread_as_not_suspended(thread);
        ready_thread(thread);
```

laurenmurphyx64, 2020-10-16:
(Note that same crash will occur regardless of whether the semaphore call is there or not.)

This code will show a page fault when logging is turned on in `prj.conf`.

`prj.conf`:
```
CONFIG_LOG=y
```

Output:
```
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200625_115407-9426dddc0a1f-zephyr
)
Booting from ROM..*** Booting Zephyr OS build zephyr-v2.4.0-634-g53cc090c39a5  ***
Main
[00:00:00.000,000] <err> os: Page fault at address 0x89fe2383 (error code 0x0)
[00:00:00.000,000] <err> os: Linear address not present in page tables
[00:00:00.000,000] <err> os: Access violation: supervisor thread not allowed to read
[00:00:00.000,000] <err> os: PDE: not present
[00:00:00.000,000] <err> os: EAX: 0x89fe2383, EBX: 0x00000002, ECX: 0x00100c14, EDX: 0x00107000
[00:00:00.000,000] <err> os: ESI: 0x00117de0, EDI: 0x00107268, EBP: 0x00117e1c, ESP: 0x00117de4
[00:00:00.000,000] <err> os: EFLAGS: 0x00000002 CS: 0x0008 CR3: 0x0011c000
[00:00:00.000,000] <err> os: call trace:
[00:00:00.000,000] <err> os: EIP: 0x00100c54
[00:00:00.000,000] <err> os:      0x00103ec0 (0x107268)
[00:00:00.000,000] <err> os:      0x00103fcc (0x107268)
[00:00:00.000,000] <err> os:      0x00104048 (0x0)
[00:00:00.000,000] <err> os:      0x001003e7 (0x107000)
[00:00:00.000,000] <err> os:      0x00103a73 (0x117e88)
[00:00:00.000,000] <err> os:      0x00100f2c (0x0)
[00:00:00.000,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.000,000] <err> os: Current thread: 0x001071e0 (unknown)
[00:00:00.000,000] <err> os: Halting system
```

PR #29299 (2020-10-22): Only resume suspended threads
- only resume suspended threads
- Do not queue a thread that is already queued

Fixes #28694
Fixes #29244
commit 081605ee (2020-10-22): kernel: do not queue a thread that is already queued

Do not add a thread to the run queue if it was already added.

Fixes #29244
closed 2020-10-22 by nashif