zephyrproject-rtos/zephyr #29258: net: Unable to establish TCP connections from Windows hosts

ohitz, 2020-10-16:
**Describe the bug**

We have problems establishing TCP connections from a Windows 10 host. This seems to have been introduced with ae139bf8a9f1d380467c5b1468997e2bd3bc3304 in https://github.com/zephyrproject-rtos/zephyr/issues/29164.

From Linux hosts, connections work fine, but not so from a Windows 10 host.

**To Reproduce**

Steps to reproduce the behavior:

1. Compile the `samples/net/sockets/echo_server/` example program and flash it to a board (tested MIMXRT1060-EVK)
2. Connect to the board from a Windows 10 host using a telnet client: `telnet <ip> 4242`, the connection seems to be hanging and the echo server doesn't log that there is a connection
3. Test the same from a Linux host and that connection works fine

When ae139bf8a9f1d380467c5b1468997e2bd3bc3304 is removed, the Windows host connects fine.

**Expected behavior**

Connections should establish no matter from where.

**Logs and console output**

Capture of a connection attempt from said Windows 10 host (at `192.168.111.25`, while the echo server is at `192.168.111.100`).
[win10-capture.zip](https://github.com/zephyrproject-rtos/zephyr/files/5389837/win10-capture.zip)

**Environment (please complete the following information):**

 - Zephyr 13682951832e7a35b4c3ebfc7eec102a478cdbb6

jukkar, 2020-10-16:
The packet flow looks a bit weird. So win10 sends SYN, but Zephyr responds with SYN too instead of SYN|ACK, then win10 sends SYN|ACK which looks really weird. It kind of looks that the connection is somehow established but because Zephyr does not receive SYN|ACK while in correct state, the accept callback is not called and the connection is not established properly.
I will investigate, there might be an easy fix.

ohitz, 2020-10-16:
Not sure if this helps, but here's a trace from the same Windows host to an embedded device running a very old version of lwip: [win10-lwip.zip](https://github.com/zephyrproject-rtos/zephyr/files/5391443/win10-lwip.zip)

ohitz, 2020-10-16:
Could it be that this check doesn't match because the ECN and CWR flags are set: https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp2.c#L1247

ohitz, 2020-10-16:
That ECN bit can also be activated on Linux: `echo 1 > /proc/sys/net/ipv4/tcp_ecn`

By doing this, Linux sets ECN|CWR on outbound connections what causes Zephyr to respond with SYN instead of SYN|ACK. The connection is established in the end, but that's probably because of how the Linux stack works.

jukkar, 2020-10-16:
Yes, it is those flags which are not handled properly. This should be an easy fix.

ohitz, 2020-10-16:
Actually, is this else clause correct at all?
https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/net/ip/tcp2.c#L1252

According to figure 6 of RFC793, there are only two ways out of the LISTEN state: either through the reception of a SYN, or through an initiated send, which, I think, wouldn't be in a function called `tcp_in()`. So why is there an else clause at all?

jukkar, 2020-10-16:
> According to figure 6 of RFC793, there are only two ways out of the LISTEN state: either through the reception of a SYN, or through an initiated send, which, I think, wouldn't be in a function called `tcp_in()`. So why is there an else clause at all?

Looks like the check to simultaneous open case
https://en.wikipedia.org/wiki/File:Tcp_state_diagram_fixed.svg

ohitz, 2020-10-16:
> Looks like the check to simultaneous open case
I don't understand this. Why would a connection which is in LISTEN state respond to anything else than a SYN? If we have a simultaneous open case, that connection is no longer in LISTEN state but has already sent a SYN, no?

jukkar, 2020-10-16:
I do not know exactly why the else is there, this is not my code. We can certainly try to remove it and see if anything breaks.

ohitz, 2020-10-19:
Thanks @jukkar your change solves the problem!

This means that before ae139bf8a9f1d380467c5b1468997e2bd3bc3304 connections requesting ECN somehow worked, but the connection setup was never really clean (tried it, but it looks more like a coincidence that they worked).

jukkar, 2020-10-19:
> This means that before [ae139bf](https://github.com/zephyrproject-rtos/zephyr/commit/ae139bf8a9f1d380467c5b1468997e2bd3bc3304) connections requesting ECN somehow worked

Yes, because the connection was accepted always and before the ACK was received from peer. That is why there was a small window where the app could not get things sent because the connection was not really up yet.

> , but the connection setup was never really clean (tried it, but it looks more like a coincidence that they worked).

To me, it looked like the received SYN packet kind of went to this "simultaneous connect" state, but cause we had already accepted the connection, everything was ok from application point of view.

PR #29277 (2020-10-19): net: tcp2: Fix TCP connection from Windows 10
Windows 10 sends ECN-Echo and Congestion Window Reduced (CWR) flags
together with SYN flag in the connection establishment but the code
did not ignore these flags and send just SYN back (instead of SYN|ACK).
This caused the connection establishement in application level to
fail as the application was never notified about it.

Fixes #29258
commit 3b6155bb (2020-10-19): net: tcp2: Fix TCP connection from Windows 10

Windows 10 sends ECN-Echo and Congestion Window Reduced (CWR) flags
together with SYN flag in the connection establishment but the code
did not ignore these flags and send just SYN back (instead of SYN|ACK).
This caused the connection establishement in application level to
fail as the application was never notified about it.

Fixes #29258
closed 2020-10-19 by jukkar