zephyrproject-rtos/zephyr #29261: boards: musca_b1: post build actions with TF-M are not done

ioannisg, 2020-10-16:
**Describe the bug**
When building Zephyr for a "non-secure" platform version and with MCUBoot and TF-M support, post-build actions involve signing the S and NS images and assembling into a single binary.

This should normally be done on the above order, but, currently, in musca_b1 board the action order is reverted. This is not consistent with the rest of the boards that are TFM capable (nRF, ST, and mps2_an521)

**To Reproduce**
See CMakeLists.txt in boards/arm/v2_musca_b1)

**Expected behavior**
We should be consistent on the post build actions across all relevant platforms, when building with TFM

**Impact**
Unknown. But LOW, definitely. Limited to musca_b1, building with TFM

ioannisg, 2020-10-16:
CC @microbuilder

galak, 2020-10-20:
@ioannisg is this just re-ordering something in `CMakeLists.txt`

closed 2021-01-03 by github-actions[bot]