zephyrproject-rtos/zephyr #29263: tests/kernel/mem_protect/obj_validation fails build on some boards after recent changes

pfalcon, 2020-10-16:
After https://github.com/zephyrproject-rtos/zephyr/pull/25641 : for boards mps2_an385, frdm_k64f, etc.:

~~~
01:39:38.262 /home/buildslave/workspace/zephyr-upstream/zephyr/tests/kernel/mem_protect/obj_validation/src/main.c: In function 'dyn_object_release':
01:39:38.262 /home/buildslave/workspace/zephyr-upstream/zephyr/tests/kernel/mem_protect/obj_validation/src/main.c:173:30: error: the comparison will always evaluate as 'true' for the address of 'test_dyn_mutex' will never be NULL [-Werror=address]
01:39:38.262   173 |  zassert_not_null(&test_dyn_mutex, "Can't allocate dynamical kobj");
01:39:38.262       |                              ^~
01:39:38.262 cc1: all warnings being treated as errors
~~~

Full sanitycheck build log: https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/5651/PLATFORM=mps2_an385,ZEPHYR_TOOLCHAIN_VARIANT=zephyr,label=docker-xenial-amd64-13/console

nashif, 2020-10-16:
@pfalcon  which toolchain are you using?

nashif, 2020-10-16:
ok, i see above you are using zephyr sdk. Not able to reproduce this locally and in Zephyr's CI, wonder why this warning is active in your environment.

This is still an issue that needs to be fixed, @maksimmasalski  you are checking the address, not the pointer.

pfalcon, 2020-10-16:
The full picture can be seen here: https://ci.linaro.org/view/lite-iot-ci/job/zephyr-upstream/5650/ . The compile warning-as-error happens synchronously for both Zephyr SDK 0.11.4 and gnuarmemb gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux.tar.bz2.

But it happens for disco_l475_iot1, frdm_k64f, mps2_an385, nucleo_f401re, and doesn't happen for cc3220sf_launchxl, frdm_kw41z, nucleo_f103rb, qemu_cortex_m3.

I have to admit I didn't try to reproduce locally - I usually do, but last couple of times I figure we were duplicating effort with a few other folks, so I decided to shoot a ticket first.

I'd also guess it's more or less clear there's a typo, in `zassert_not_null(&test_dyn_mutex, "Can't allocate dynamical kobj");` , the `&` looks stray (unless there's a bigger logic error in context).

PR #29291 (2020-10-16): Revert "tests: add new kernel objects tests"
This test is generating build warnings as it is making
checks that can never be false.

This reverts commit a4f1a5f58f61820526560d4867eceafc81175a07.

Fixes: #29263
closed 2020-10-16 by nashif