zephyrproject-rtos/zephyr #29274: Can't get Coded PHY type(S2 or S8)

ustinov-dev, 2020-10-16:
Hi,
I use Zephyr from the master branch and struggle with Coded PHY type detection in created connection.
I can't find anything except `bt_conn_get_info` to get the PHY type, but unfortunately the `bt_conn_le_phy_info` structure from this function doesn't contain any information about S2 or S8, only Coded or not.

Is there any opportunity to get all information about used PHY?

Thanks in advance,
Sergey

joerchan, 2020-10-20:
When LE Coded PHY is used then both S2 and S8 can be used, the connection can switch coding per PDU sent.

ustinov-dev, 2020-10-21:
Shame on me =(
I didn't know about it.

I try to understand a case when I have a peripheral device(nRf52840), that get PHY update request from the Central(nRf52840) and update own PHY after it.
I found only ` void (*le_phy_updated)(struct bt_conn *conn, struct bt_conn_le_phy_info *param)` callback and` bt_conn_get_info` function, that can return `bt_conn_le_phy_info`.
But` bt_conn_le_phy_info` can't get default or current tx Code PHY type(I mean S2 or S8).

1) Is there any opportunity to check PHY update request parameters before setting, or only after?
2) Is it possible to get default or current tx Coded PHY type(S2 or S8) for the connection?

joerchan, 2020-10-22:
@ustinov-dev You cannot know if S2 or S8 is used, you can only know that LE Coded PHY is used, and then both S2 and S8 will be used.

1. No there is no way in zephyr to get PHY update request parameters (if you have used softdevice before there might be something like that, but that is not here).
2. No, but you can set it using bt_conn_le_phy_update.

ustinov-dev, 2020-10-22:
Ok. I see.

Thanks a lot guys!

closed 2020-10-22 by ustinov-dev

Arjun765, 2021-11-02:
>you can only know that LE Coded PHY is used, and then both S2 and S8 will be used.

Will the controller automatically switch between the two?

joerchan, 2021-11-02:
> Will the controller automatically switch between the two?

The controller is free to do that. Behavior will vary depending on the implementation.