zephyrproject-rtos/zephyr #29284: compilation issues for MinnowBoard/ UpSquared on documentation examples

plbossart, 2020-10-16:
**Describe the bug**

Following the documentation and trying to compile for minnowboard and up_squared fails.

Either the documentation is missing steps, or the code is missing configurations

**To Reproduce**

Install with the instructions in https://docs.zephyrproject.org/latest/getting_started/index.html and use -b minnowboard or -b up_squared

````
pbossart@pbossart-mobl3:/data/pbossart/zephyrproject/zephyr$ west build -p auto -b minnowboard samples/basic/blinky
-- west build: generating a build system
Including boilerplate (Zephyr base): /data/pbossart/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /data/pbossart/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.99 (/data/pbossart/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: minnowboard
-- Cache files will be written to: /home/pbossart/.cache/zephyr
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/usr/local/zephyr-sdk-0.11.4)
-- Found dtc: /usr/local/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /data/pbossart/zephyrproject/zephyr/boards/x86/minnowboard/minnowboard.dts
-- Generated zephyr.dts: /data/pbossart/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /data/pbossart/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /data/pbossart/zephyrproject/zephyr/Kconfig
Loaded configuration '/data/pbossart/zephyrproject/zephyr/boards/x86/minnowboard/minnowboard_defconfig'
Merged configuration '/data/pbossart/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/data/pbossart/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/data/pbossart/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
-- Configuring done
CMake Error at ../../../cmake/extensions.cmake:398 (add_library):
  No SOURCES given to target: drivers__gpio
Call Stack (most recent call first):
  ../../../cmake/extensions.cmake:375 (zephyr_library_named)
  ../../../drivers/gpio/CMakeLists.txt:3 (zephyr_library)

CMake Generate step failed.  Build files cannot be regenerated correctly.
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/data/pbossart/zephyrproject/zephyr/build -S/data/pbossart/zephyrproject/zephyr/samples/basic/blinky -GNinja -DBOARD=minnowboard
````

````
pbossart@pbossart-mobl3:/data/pbossart/zephyrproject/zephyr$ west build -p auto -b up_squared samples/basic/blinky
-- west build: making build dir /data/pbossart/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /data/pbossart/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /data/pbossart/zephyrproject/zephyr/samples/basic/blinky
-- Zephyr version: 2.4.99 (/data/pbossart/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.8.0", minimum required is "0.7.1")
-- Board: up_squared
-- Cache files will be written to: /home/pbossart/.cache/zephyr
ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found toolchain: zephyr (/usr/local/zephyr-sdk-0.11.4)
-- Found dtc: /usr/local/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /data/pbossart/zephyrproject/zephyr/boards/x86/up_squared/up_squared.dts
-- Generated zephyr.dts: /data/pbossart/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /data/pbossart/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /data/pbossart/zephyrproject/zephyr/Kconfig
Loaded configuration '/data/pbossart/zephyrproject/zephyr/boards/x86/up_squared/up_squared_defconfig'
Merged configuration '/data/pbossart/zephyrproject/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/data/pbossart/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/data/pbossart/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /data/pbossart/zephyrproject/zephyr/build
-- west build: building application
[1/113] Preparing syscall dependency handling

[14/113] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /usr/local/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/x86_64-zephyr-elf-gcc -DBUILD_VERSION=zephyr-v2.4.0-679-gea2107c4295e -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR__=1 -I/data/pbossart/zephyrproject/zephyr/include -Izephyr/include/generated -I/data/pbossart/zephyrproject/zephyr/soc/x86/apollo_lake -isystem /data/pbossart/zephyrproject/zephyr/lib/libc/minimal/include -isystem /usr/local/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/include -isystem /usr/local/zephyr-sdk-0.11.4/x86_64-zephyr-elf/bin/../lib/gcc/x86_64-zephyr-elf/9.2.0/include-fixed -m64    -Os -imacros /data/pbossart/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -m64 -Wa,--divide -imacros /data/pbossart/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-address-of-packed-member -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/data/pbossart/zephyrproject/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=/data/pbossart/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/data/pbossart/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -march=goldmont -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c /data/pbossart/zephyrproject/zephyr/samples/basic/blinky/src/main.c
/data/pbossart/zephyrproject/zephyr/samples/basic/blinky/src/main.c:24:2: error: #error "Unsupported board: led0 devicetree alias is not defined"
   24 | #error "Unsupported board: led0 devicetree alias is not defined"
      |  ^~~~~
[23/113] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/onoff.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /data/pbossart/zephyrproject/zephyr/build
````

**Expected behavior**

Examples compile for "supported boards"

**Impact**

Minor, likely documentation or small configuration issue.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...) zephyr-sdk-0.11.4:
 - Commit SHA or Version used ea2107c4295e78e5733f78be3e1b8ccceaf28430

nashif, 2020-10-19:
not every sample is required to build and run on every board, there are perquisites expressed in the sample.yaml file, in this case we need the definition of the LED which we do not have for those boards.

nashif, 2020-12-11:
closing per comment above

closed 2020-12-11 by nashif