zephyrproject-rtos/zephyr #29295: Improved user interface for devicetree (like guiconfig)

mbolivar-nordic, 2020-10-16:
**Is your enhancement proposal related to a problem? Please describe.**

Devicetree is widely used by Zephyr, but it lacks convenient tooling for development.

It is used by most in-tree device drivers to instantiate `struct device` objects and describe desired boot-time configuration. Users are expected to understand enough about devicetree to configure and enable devices using devicetree overlays, and an increasing amount of basic pin control is using devicetree as well.

However, the user-visible tooling for viewing and editing the final devicetree is lacking in maturity compared to Kconfig. Users basically have to open `build/zephyr/zephyr.dts` and `build/zephyr/include/generated/devicetree_unfixed.h` to get information about the final devicetree itself and check their work while developing.

This is a usability hurdle which also causes support burden. The complexity of dealing with devicetree is a regular complaint against the project.

**Describe the solution you'd like**

A graphical devicetree viewer and editor, like 'guiconfig' for DTS. This program should be able to load the devicetree from a build directory and pair nodes with contextual information about bindings, as well as suggest or even perform edits (such as "here's how to refer to this property from C", or "here's how to enable this node in an overlay file").

**Describe alternatives you've considered**

A curses-based console application like 'menuconfig' was considered but rejected because:

1. I am not willing to write one, and anyway
2. the [windows-curses](https://github.com/zephyrproject-rtos/windows-curses) library we depend on for cross-platform curses is going unmaintained

**Additional context**

I am inspired by the excellent work done in @trond-snekvik 's [VS code extension](https://marketplace.visualstudio.com/items?itemName=trond-snekvik.devicetree) and @ulfalizer 's 'menuconfig' and 'guiconfig', as well as the original Linux C programs for managing the kconfig tree.

I would like to have a portable, IDE-independent interface for Zephyr that uses edtlib and is written in Python.

mbolivar-nordic, 2020-10-16:
No schedule and I have not started work. I'm just putting this into the world as something I want to do to get comments from elsewhere.

gmarull, 2020-10-21:
I can offer some help if PySide2 (a.k.a. Qt for Python) is considered as a viable GUI framework.

mbolivar-nordic, 2020-10-23:
> I can offer some help if PySide2 (a.k.a. Qt for Python) is considered as a viable GUI framework.

That's exactly what I'm thinking. Let's talk when it comes time.

ExtremeGTX, 2020-10-25:
> I can offer some help if PySide2 (a.k.a. Qt for Python) is considered as a viable GUI framework.

Does it require installation of Qt5 ? 
Why not Tkinter which is bundled with `python3` also it is used in `guiconfig`.

mbolivar-nordic, 2020-10-26:
> Does it require installation of Qt5 ?

You can install PySide2 it with pip just like any other binary python dependency.

> Why not Tkinter which is bundled with `python3` also it is used in `guiconfig`.

To be honest, because tkinter based interfaces look really dated and there doesn't seem to be a lot of developer momentum behind the project.

mbolivar-nordic, 2021-02-05:
I ran across this interesting message on the DT spec mailing list:

https://www.spinics.net/lists/devicetree-spec/msg00950.html

It seems to be exactly what this issue anticipates and something to investigate further. cc @trond-snekvik

rlneumiller, 2024-11-24:
I personally would find this to be quite useful to help with understanding how to use dt.