zephyrproject-rtos/zephyr #29307: samples/bluetooth/mesh-demo unable to send vendor button message

robintschroeder, 2020-10-18:
Running the zephyr bluetooth mesh sample from master branch on a BBC micro:bit, I get the following error when I press the button. 
Seems the vendor button message is not being sent.  
Any help would be appreciated! 

`*** Booting Zephyr OS build zephyr-v2.4.0-685-g7e64fba9ed84  ***
Initializing...
Unicast address: 0x0001
Bluetooth initialized
Mesh initialized
Loading stored settings
Using stored settings
Subscribing to heartbeat messages
button_send_pressed()
**Unable to send Vendor Button message**
Button message sent with OpCode 0x00c005f1'

robintschroeder, 2020-10-22:
I have been tinkering with this a bit. 
In prj_bbc_microbit.conf, if I set CONFIG_BT_SETTINGS=n instead of y, the micro:bit provisions successfully. 
However, then I start getting -12 errors when binding, publishing and subscribing to models, which seems to mean that I am out of memory.
#define ENOMEM          12              /* Cannot allocate memory */

So, seems like past versions of Zephyr were a little smaller and fit nicely on the limited memory of a micro:bit, but these later versions are too big? Should I be assuming that?? 
Any guidance would be greatly appreciated! Thanks in advance!

robintschroeder, 2020-10-28:
Rolled back to Zephyr 1.14.
The example isn't fully working property, but it runs and in part, does what the demo intends.

WilliamGFish, 2020-11-05:
Rolling back 1.14 LTS is the best option as there are things in play here:
- The rewrite of BLE subsystem struggles with the m:bit memory constraints
- The nRF51 series in being phased out
- A new Micro:bit has just been released based on the nRF52833 (Nov 2020)

I expect in a matter of weeks there will support in zephyr for the latest Micro:bit

robintschroeder, 2020-11-06:
Thanks for your comment!

closed 2020-11-06 by robintschroeder